alibaba/tidevice #121: 通过threading.thread启动监控后报错：ssl.SSLEOFError: EOF occurred in violation of protocol (_ssl.c:2423)

hhhzhao, 2021-07-16:
首先感谢大神！
现在有一个问题就是 我想写一个可以同时监控多台ios设备的程序，所以就想到了用threading.Thread方法
这是开启方法：
def performance_switch_on(self, package):
       self.package = package
       self.device_performance.start(self.package, callback=self.callback)
            while self.switch:
                 time.sleep(5)

在用线程起的时候(threading.Thread(target=self.ios.performance_switch_on, args=(package, )).start()会遇到如下的报错：
Exception in thread Thread-7:
Traceback (most recent call last):
  File "C:\Python3\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Python3\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Python3\lib\site-packages\tidevice\_instruments.py", line 560, in _drain
    self._drain_single_message()
  File "C:\Python3\lib\site-packages\tidevice\_instruments.py", line 611, in _drain_single_message
    self._reply_null(dtxm)
  File "C:\Python3\lib\site-packages\tidevice\_instruments.py", line 518, in _reply_null
    message_id=m.message_id)
  File "C:\Python3\lib\site-packages\tidevice\_instruments.py", line 392, in send_dtx_message
    self.sendall(data)
  File "C:\Python3\lib\site-packages\tidevice\_safe_socket.py", line 66, in sendall
    return self._sock.sendall(data)
  File "C:\Python3\lib\ssl.py", line 1034, in sendall
    v = self.send(byte_view[count:])
  File "C:\Python3\lib\ssl.py", line 1003, in send
    return self._sslobj.write(data)
ssl.SSLEOFError: EOF occurred in violation of protocol (_ssl.c:2423)

有什么比较好的解决方法呢？

huang1988519, 2021-07-20:
指定 tidevice perf 后报错

huang1988519, 2021-07-22:
> 指定 tidevice perf 后报错
> 
> `Exception in thread Thread-3: Traceback (most recent call last): File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/threading.py", line 932, in _bootstrap_inner self.run() File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/threading.py", line 870, in run self._target(*self._args, **self._kwargs) File "/Users/hwh/Library/Python/3.8/lib/python/site-packages/tidevice/_instruments.py", line 560, in _drain self._drain_single_message() File "/Users/hwh/Library/Python/3.8/lib/python/site-packages/tidevice/_instruments.py", line 611, in _drain_single_message self._reply_null(dtxm) File "/Users/hwh/Library/Python/3.8/lib/python/site-packages/tidevice/_instruments.py", line 516, in _reply_null self.send_dtx_message(m.channel_id, File "/Users/hwh/Library/Python/3.8/lib/python/site-packages/tidevice/_instruments.py", line 392, in send_dtx_message self.sendall(data) File "/Users/hwh/Library/Python/3.8/lib/python/site-packages/tidevice/_safe_socket.py", line 66, in sendall return self._sock.sendall(data) File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/ssl.py", line 1204, in sendall v = self.send(byte_view[count:]) File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/ssl.py", line 1173, in send return self._sslobj.write(data) OSError: [Errno 22] Invalid argument`

此问题已修复，提交 PR，等待合并即可

jianghuihon, 2025-06-10:
2025-05-28 07:14:42,700 - INFO - tools.py:783 - AutoTest - tidevice -u a6931846e697a75906d9610f8fb65d8218c1ea74 ps ret:Traceback (most recent call last):
  File "/Users/v_gsyou/Library/Python/3.9/bin/tidevice", line 8, in <module>
    sys.exit(main())
  File "/Users/v_gsyou/Library/Python/3.9/lib/python/site-packages/tidevice/__main__.py", line 1027, in main
    actions[args.subparser](args)
  File "/Users/v_gsyou/Library/Python/3.9/lib/python/site-packages/tidevice/__main__.py", line 619, in cmd_ps
    with d.connect_instruments() as ts:
  File "/Users/v_gsyou/Library/Python/3.9/lib/python/site-packages/decorator.py", line 232, in fun
    return caller(func, *(extras + args), **kw)
  File "/Users/v_gsyou/Library/Python/3.9/lib/python/site-packages/retry/api.py", line 73, in retry_decorator
    return __retry_internal(partial(f, *args, **kwargs), exceptions, tries, delay, max_delay, backoff, jitter,
  File "/Users/v_gsyou/Library/Python/3.9/lib/python/site-packages/retry/api.py", line 33, in __retry_internal
    return f()
  File "/Users/v_gsyou/Library/Python/3.9/lib/python/site-packages/tidevice/_device.py", line 817, in connect_instruments
    conn = self.start_service(LockdownService.InstrumentsRemoteServer)
  File "/Users/v_gsyou/Library/Python/3.9/lib/python/site-packages/tidevice/_device.py", line 501, in start_service
    return self._unsafe_start_service(name)
  File "/Users/v_gsyou/Library/Python/3.9/lib/python/site-packages/tidevice/_device.py", line 509, in _unsafe_start_service
    with self.create_session() as _s:
  File "/Users/v_gsyou/Library/Python/3.9/lib/python/site-packages/tidevice/_device.py", line 336, in create_session
    s.psock.switch_to_ssl(self.ssl_pemfile_path)
  File "/Users/v_gsyou/Library/Python/3.9/lib/python/site-packages/tidevice/_safe_socket.py", line 171, in switch_to_ssl
    ssock = context.wrap_socket(self._sock, server_hostname="iphone.localhost")
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/ssl.py", line 500, in wrap_socket
    return self.sslsocket_class._create(
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/ssl.py", line 1040, in _create
    self.do_handshake()
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLEOFError: EOF occurred in violation of protocol (_ssl.c:1129)