alibaba/tidevice #124: fsync功能失效命令如下：tidevice fsync -B BUNDLEID ls ./

wangjiawei9816, 2021-08-04:
[image]

Traceback (most recent call last):
  File "F:/MyCode/python/taobao-iphone-device-main/tidevice/__main__.py", line 720, in <module>
    main()
  File "F:/MyCode/python/taobao-iphone-device-main/tidevice/__main__.py", line 715, in main
    actions[args.subparser](args)
  File "F:/MyCode/python/taobao-iphone-device-main/tidevice/__main__.py", line 404, in cmd_fsync
    pprint(sync.listdir(arg0))
  File "F:\MyCode\python\taobao-iphone-device-main\tidevice\_sync.py", line 106, in listdir
    pkg = self._recv()
  File "F:\MyCode\python\taobao-iphone-device-main\tidevice\_sync.py", line 75, in _recv
    self.recvall(4) # Discard plist xml-content
  File "F:\MyCode\python\taobao-iphone-device-main\tidevice\_safe_socket.py", line 61, in recvall
    raise MuxError("socket connection broken")
exceptions.MuxError: socket connection broken

wangjiawei9816, 2021-08-04:
多人都有此问题，疑似协议不匹配，麻烦官方的人验证一下：
window系统下，使用tidevice的fsync进行访问指定bundle的任何文件夹，都会报错

wangjiawei9816, 2021-08-05:
还是一样的报错，并且多人复现在ios14.5以上的版本

wangjiawei9816, 2021-08-05:
https://github.com/alibaba/taobao-iphone-device/issues/100 同样的问题，可能是magic失效了

codeskyblue, 2021-08-05:
你们的好像不太一样，我现在手机还没升级到14.5..  App的list需要App设置了 可以访问沙盒的权限才可以

wangjiawei9816, 2021-08-05:
得在windows系统下，你是不是在mac机上的，，

codeskyblue, 2021-08-05:
是的，我用的Mac M1

wangjiawei9816, 2021-08-05:
你可以尝试一下windows机器，必定报错，我们很多用户都是因为它能在Windows平台使用才选择tidevice的，如果可以，请帮忙查一下问题，我也可以帮忙辅助查问题

wangjiawei9816, 2021-08-05:
可以参考这个地方修改，_device.py搜索vendcontainer改成venddocuments即可
https://github.com/GNOME/gvfs/commit/b8ad223b1e2fbe0aec24baeec224a76d91f4ca2f

codeskyblue, 2021-08-06:
Fixed in tidevice 0.4.15. Thanks @wangjiawei9816

closed 2021-08-06 by codeskyblue

wangjiawei9816, 2021-08-06:
直接改会有一个问题，listdir（‘/tmp’）失败，得区分不同情况

commit fff6a9ac (2021-08-06): fix fsync ls app error, close #124, ref libimobiledevice/libimobiledevice#193
commit c0c63d04 (2021-08-12): fix fsync ls app error, close #124, ref libimobiledevice/libimobiledevice#193