alibaba/tidevice #130: tidevice launch 命令传递参数

SZDXDZXXXJ, 2021-08-19:
我想使用launch命令唤起APP并打开指定页面，使用tidevice launch -h查看命令使用方法，发现可以传递app arguments
tidevice launch -h
usage: tidevice launch [-h] [--kill] bundle_id [arguments [arguments ...]]

positional arguments:
  bundle_id   app bundleId
  arguments   app arguments (default: None)

optional arguments:
  -h, --help  show this help message and exit
  --kill      kill app if running (default: False)
自己尝试了下，tidevice luanch bundle_id  123（123替换为APP内的页面地址）
能正常唤起APP，但是获取不到参数，请问应该如何正确使用？

codeskyblue, 2021-08-20:
我知道这个功能是没什么问题，因为拿别的app启动测试过

SZDXDZXXXJ, 2021-08-22:
> 我知道这个功能是没什么问题，因为拿别的app启动测试过

具体的命令可以发我参考下么？我怀疑是我命令写的不对，感谢！

enigma9527, 2022-01-06:
> 

你好，请问这个命令要怎么写？谢谢

LXManMan, 2022-11-03:
怎么传参数都不对，cmd 命令行怎么写有知道的吗，还是说无法传入参数，不走launch的通道

LXManMan, 2022-11-04:
已验证
tidevice launch xxx 测试tidevice传参 
可以传参，在app 的launch方法中获取进程信息，argruments 第二个开始就是我们需要的参数