alibaba/tidevice #132: ssl.SSLEOFError: EOF occurred in violation of protocol (_ssl.c:2483)

JennyPan2016, 2021-08-19:
采集时异常：
[image]

Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/local/Cellar/python@3.9/3.9.6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/usr/local/Cellar/python@3.9/3.9.6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/site-packages/tidevice/_instruments.py", line 560, in _drain
    self._drain_single_message()
  File "/usr/local/lib/python3.9/site-packages/tidevice/_instruments.py", line 611, in _drain_single_message
    self._reply_null(dtxm)
  File "/usr/local/lib/python3.9/site-packages/tidevice/_instruments.py", line 516, in _reply_null
    self.send_dtx_message(m.channel_id,
  File "/usr/local/lib/python3.9/site-packages/tidevice/_instruments.py", line 392, in send_dtx_message
    self.sendall(data)
  File "/usr/local/lib/python3.9/site-packages/tidevice/_safe_socket.py", line 66, in sendall
    return self._sock.sendall(data)
  File "/usr/local/Cellar/python@3.9/3.9.6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/ssl.py", line 1204, in sendall
    v = self.send(byte_view[count:])
  File "/usr/local/Cellar/python@3.9/3.9.6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/ssl.py", line 1173, in send
    return self._sslobj.write(data)
ssl.SSLEOFError: EOF occurred in violation of protocol (_ssl.c:2483)

codeskyblue, 2021-08-20:
tidevice版本多少？

JennyPan2016, 2021-08-22:
更新到最新的tidevice版本后没有报错了。

closed 2021-08-22 by JennyPan2016

pxrgod, 2021-09-08:
> 更新到最新的tidevice版本后没有报错了。

更新到0.4.18版本上出现了这种报错