alibaba/tidevice #133: 自己通过xcodebuild build-for-testing命令打包出来的ipa时间一长再次启动tidevice xctest -B就会报错，重新打包才行，这是什么原因呢？有什么办法能解决吗？

jeniva, 2021-08-20:
我试过几个App和证书都是有这样的问题，这是什么原因？要怎么解决呢？
F:\Autotest\Add_devices>tidevice xctest -B xxx.xxx.xctrunner
[I 210813 15:53:01 _device:964] BundleID: xxx.xxx.xctrunner
[I 210813 15:53:01 _device:967] ProductVersion: 11.4.1
[I 210813 15:53:01 _device:969] DeviceIdentifier: xxxxxxxxxxxxxxx
[I 210813 15:53:01 _device:808] SignIdentity: 'Apple Development: xxxxxxxxxxxx'
[I 210813 15:53:01 _device:814] CFBundleExecutable: xxxxxx-Runner
[I 210813 15:53:01 _device:845] AppContainer: /private/var/mobile/Containers/Data/Application/207159FE-F718-48F6-BC8F-2EC1DFE348C7
[E 210813 15:53:02 _device:882] Launch failed: NSError(CODE:1 DOMAIN:DTXMessage INFO:{'DTXExceptionKey': 'The operation couldn’t be completed. '
'(FBSOpenApplicationErrorDomain error 1.) : Failed to '
'launch process with bundle identifier '
"'hikvison.FastbotRunner.xctrunner'"})
Traceback (most recent call last):
File "c:\python37\lib\runpy.py", line 193, in _run_module_as_main
"__main__", mod_spec)
File "c:\python37\lib\runpy.py", line 85, in _run_code
exec(code, run_globals)
File "C:\python37\Scripts\tidevice.exe\__main__.py", line 9, in <module>
File "c:\python37\lib\site-packages\tidevice\__main__.py", line 720, in main
actions[args.subparser](args)
File "c:\python37\lib\site-packages\tidevice\__main__.py", line 226, in cmd_xctest
env=env)
File "c:\python37\lib\site-packages\tidevice\_device.py", line 1042, in xctest
env=env, logger=xclogger)
File "c:\python37\lib\site-packages\tidevice\_device.py", line 883, in _launch_app_runner
raise MuxError("Launch failed")
tidevice.exceptions.MuxError: Launch failed
[image]

codeskyblue, 2021-08-20:
证书过期了吧

jeniva, 2021-08-20:
我用的开发的证书，证书本身还没有过期的，我用这个证书重新打包也还是可以安装的，就是之前打包出来后的wda的ipa包会过期，我不知道是有什么参数设置吗还是啥的？你们build出来的ipa可以一直用的吗？

chenhengjie123, 2022-03-23:
建议先确认下打包证书选择的对不对。现象上看像是证书过期。

默认免费的真机调试证书，有效期只有7天，7天后自动过期，过期后无法安装或启动。每次重新打包会重新生成一个新的证书（有效期是打包时开始算起的7天）。只有付费的有效期才长。

如果你本地既有免费的又有付费的（比如有不止一个 apple account 在 xcode 里登录），会容易搞混，建议是退出所有非付费的账号，避免搞混。