alibaba/tidevice #134: tidevice ps多次执行时，会出现tidedvice.exception.MuxError

pxrgod, 2021-08-24:
Traceback (most recent call last):
  File "/opt/homebrew/bin/tidevice", line 8, in <module>
    sys.exit(main())
  File "/Users/hspcadmin/Library/Python/3.9/lib/python/site-packages/tidevice/__main__.py", line 723, in main
    actions[args.subparser](args)
  File "/Users/hspcadmin/Library/Python/3.9/lib/python/site-packages/tidevice/__main__.py", line 441, in cmd_ps
    ps = list(d.instruments.app_process_list(app_infos))
  File "/Users/hspcadmin/Library/Python/3.9/lib/python/site-packages/tidevice/_instruments.py", line 708, in app_process_list
    processes = self.app_running_processes()
  File "/Users/hspcadmin/Library/Python/3.9/lib/python/site-packages/tidevice/_instruments.py", line 681, in app_running_processes
    retobj = self.call_message(identifier, "runningProcesses")
  File "/Users/hspcadmin/Library/Python/3.9/lib/python/site-packages/tidevice/_instruments.py", line 354, in call_message
    channel = self.make_channel(channel)
  File "/Users/hspcadmin/Library/Python/3.9/lib/python/site-packages/tidevice/_instruments.py", line 315, in make_channel
    result = self.call_message(0, '_requestChannelWithCode:identifier:', aux)
  File "/Users/hspcadmin/Library/Python/3.9/lib/python/site-packages/tidevice/_instruments.py", line 360, in call_message
    return self.wait_reply(_id).result
  File "/Users/hspcadmin/Library/Python/3.9/lib/python/site-packages/tidevice/_instruments.py", line 550, in wait_reply
    raise MuxError("connection closed")
tidevice.exceptions.MuxError: connection closed

yipainbingxin, 2022-03-23:
请问现在有解决么，我也遇到一样的问题

codeskyblue, 2022-03-23:
我也没招，遇到这种情况，我就直接重启手机

JayGitH, 2022-04-07:
请问现在有解决么，我也遇到一样的问题
@codeskyblue