alibaba/tidevice #135: tidevice install失败了 版本tidevice version 0.4.18

hc2088, 2021-08-24:
- CreatingStagingDirectory (5%)
- ExtractingPackage (15%)
[E 210824 17:27:36 _device:744] PackageExtractionFailed
[E 210824 17:27:36 _device:745] Could not extract archive
Traceback (most recent call last):
  File "/usr/local/bin/tidevice", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/site-packages/tidevice/__main__.py", line 723, in main
    actions[args.subparser](args)
  File "/usr/local/lib/python3.9/site-packages/tidevice/__main__.py", line 128, in cmd_install
    bundle_id = d.app_install(args.filepath_or_url)
  File "/usr/local/lib/python3.9/site-packages/tidevice/_device.py", line 746, in app_install
    raise ServiceError(progress['Error'])
tidevice.exceptions.ServiceError: PackageExtractionFailed

chengducode, 2021-11-24:
请问下解决了吗 我也遇到了相同的问题

codeskyblue, 2022-05-09:
方便的话提供一下安装包

closed 2022-05-09 by codeskyblue

gangeyubo, 2022-08-01:
可能是设备上的存储空间不足了，清一下空间解决了