alibaba/tidevice #138: tidevice install报错ApplicationVerificationFailed

hc2088, 2021-09-02:
- CreatingStagingDirectory (5%)
- ExtractingPackage (15%)
- InspectingPackage (20%)
- TakingInstallLock (20%)
- PreflightingApplication (30%)
- InstallingEmbeddedProfile (30%)
- VerifyingApplication (40%)
[E 210902 14:41:18 _device:744] ApplicationVerificationFailed
[E 210902 14:41:18 _device:745] Failed to verify code signature of /var/installd/Library/Caches/com.apple.mobile.installd.staging/temp.p4l0K1/extracted/Payload/test.app : 0xe8008016 (The executable was signed with invalid entitlements.)
Traceback (most recent call last):
  File "/usr/local/bin/tidevice", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/site-packages/tidevice/__main__.py", line 723, in main
    actions[args.subparser](args)
  File "/usr/local/lib/python3.9/site-packages/tidevice/__main__.py", line 128, in cmd_install
    bundle_id = d.app_install(args.filepath_or_url)
  File "/usr/local/lib/python3.9/site-packages/tidevice/_device.py", line 746, in app_install
    raise ServiceError(progress['Error'])
tidevice.exceptions.ServiceError: ApplicationVerificationFailed

geekmister, 2022-04-20:
我前几天也遇到这个问题了，跟你错误信息一样，看你的问题提出的挺久的了，请问现在解决了？

codeskyblue, 2022-04-24:
这个是应用包的问题，不是tidevice工具的问题