alibaba/tidevice #139: tidevice自动化时，执行了tidevice wdaproxy命令后，再使用wda库跑自动化，跑一小段时间就退出了，这个是怎么回事？

pxrgod, 2021-09-07:
C:\Users\panxr33970>tidevice wdaproxy -B com.facebook.WebDriverAgentRunner2.xctrunner --port 8200
[I 210907 19:26:07 _wdaproxy:125] [00008030-000E15143A7B802E] WDA check every 30.0 seconds
[D 210907 19:26:07 _wdaproxy:131] [00008030-000E15143A7B802E] launch WDA
[I 210907 19:26:09 _device:964] BundleID: com.facebook.WebDriverAgentRunner2.xctrunner
[I 210907 19:26:09 _device:967] ProductVersion: 14.0.1
[I 210907 19:26:09 _device:969] DeviceIdentifier: 00008030-000E15143A7B802E
[I 210907 19:26:09 _device:808] SignIdentity: 'Apple Development: Yufeng Zheng (N5YJ4VFY42)'
[I 210907 19:26:09 _device:814] CFBundleExecutable: WebDriverAgentRunner-Runner
[I 210907 19:26:09 _device:845] AppContainer: /private/var/mobile/Containers/Data/Application/9E5D4901-35F1-4227-81EF-F57A6E729FD8
[I 210907 19:26:09 _device:885] Launch 'com.facebook.WebDriverAgentRunner2.xctrunner' pid: 89933
[I 210907 19:26:09 _device:1018] Test runner ready detected
[I 210907 19:26:09 _device:1010] Start execute test plan with IDE version: 29
[I 210907 19:26:10 _device:921] WebDriverAgent start successfully
Exception in thread Thread-2[I 210907 19:26:47 _device:1090]:
 xctrunner quited
Traceback (most recent call last):
  File "e:\python 3.9.0\lib\threading.py", line 950, in _bootstrap_inner
[I 210907 19:26:47 _wdaproxy:117] [00008030-000E15143A7B802E] WDA keeper stopped
[I 210907 19:26:47 _wdaproxy:158] [00008030-000E15143A7B802E] WDA stopped for the 1st time, running 0.6 minutes
[D 210907 19:26:47 _wdaproxy:131] [00008030-000E15143A7B802E] launch WDA

pxrgod, 2021-09-08:
windows环境和mac机环境都一样，tidevice版本为0.4.18,wad版本是1.4.3

SuperChaoM, 2021-09-13:
app_env = {
            'CA_ASSERT_MAIN_THREAD_TRANSACTIONS': '0',
            'CA_DEBUG_TRANSACTIONS': '0',
            'DYLD_FRAMEWORK_PATH': app_path + '/Frameworks:',
            'DYLD_LIBRARY_PATH': app_path + '/Frameworks',
            'NSUnbufferedIO': 'YES',
            'SQLITE_ENABLE_THREAD_ASSERTIONS': '1',
            'WDA_PRODUCT_BUNDLE_IDENTIFIER': '',
#            'XCTestConfigurationFilePath': xctestconfiguration_path,
            'XCTestConfigurationFilePath':'',
            'XCODE_DBG_XPC_EXCLUSIONS': 'com.apple.dt.xctestSymbolicator',
            'SWIFTUI_VIEW_DEBUG':287,
#            'XCTestSessionIdentifier':str.upper(str(session_identifier)),
            'XCTestBundlePath': f"{app_info['Path']}/PlugIns/{target_name}.xctest",
             '__XCODE_BUILT_PRODUCTS_DIR_PATHS': '/tmp/derivedDataPath/Build/Products/Release-iphoneos',
             '__XPC_DYLD_FRAMEWORK_PATH': '/tmp/derivedDataPath/Build/Products/Release-iphoneos',
             '__XPC_DYLD_LIBRARY_PATH': '/tmp/derivedDataPath/Build/Products/Release-iphoneos',
            'MJPEG_SERVER_PORT': '',
            'USE_PORT': '',
            # maybe no needed
            'LLVM_PROFILE_FILE': app_container + "/tmp/%p.profraw", # %p means pid
        } # yapf: disable
你修改_device.py代码中env 看看是否还掉线

zhixideyu, 2021-09-28:
> ### app_env = {
> ```
>         'CA_ASSERT_MAIN_THREAD_TRANSACTIONS': '0',
>         'CA_DEBUG_TRANSACTIONS': '0',
>         'DYLD_FRAMEWORK_PATH': app_path + '/Frameworks:',
>         'DYLD_LIBRARY_PATH': app_path + '/Frameworks',
>         'NSUnbufferedIO': 'YES',
>         'SQLITE_ENABLE_THREAD_ASSERTIONS': '1',
>         'WDA_PRODUCT_BUNDLE_IDENTIFIER': '',
>         'XCTestConfigurationFilePath':'',
>         'XCODE_DBG_XPC_EXCLUSIONS': 'com.apple.dt.xctestSymbolicator',
>         'SWIFTUI_VIEW_DEBUG':287,
>         'XCTestBundlePath': f"{app_info['Path']}/PlugIns/{target_name}.xctest",
>          '__XCODE_BUILT_PRODUCTS_DIR_PATHS': '/tmp/derivedDataPath/Build/Products/Release-iphoneos',
>          '__XPC_DYLD_FRAMEWORK_PATH': '/tmp/derivedDataPath/Build/Products/Release-iphoneos',
>          '__XPC_DYLD_LIBRARY_PATH': '/tmp/derivedDataPath/Build/Products/Release-iphoneos',
>         'MJPEG_SERVER_PORT': '',
>         'USE_PORT': '',
>         # maybe no needed
>         'LLVM_PROFILE_FILE': app_container + "/tmp/%p.profraw", # %p means pid
>     } # yapf: disable
> ```
> 
> 你修改_device.py代码中env 看看是否还掉线

你自己试了有用？

pxrgod, 2021-09-30:
我最后没有使用tidevice，而是使用传统的wda启动方式来做的发自我的华为手机-------- 原始邮件 --------发件人： 窒息得鱼 ***@***.***>日期： 2021年9月28日周二 晚上10:12收件人： alibaba/taobao-iphone-device ***@***.***>抄送： pxrgod ***@***.***>, Author ***@***.***>主    题： Re: [alibaba/taobao-iphone-device] tidevice自动化时，执行了tidevice wdaproxy命令后，再使用wda库跑自动化，跑一小段时间就退出了，这个是怎么回事？ (#139)

app_env = {
        'CA_ASSERT_MAIN_THREAD_TRANSACTIONS': '0',
        'CA_DEBUG_TRANSACTIONS': '0',
        'DYLD_FRAMEWORK_PATH': app_path + '/Frameworks:',
        'DYLD_LIBRARY_PATH': app_path + '/Frameworks',
        'NSUnbufferedIO': 'YES',
        'SQLITE_ENABLE_THREAD_ASSERTIONS': '1',
        'WDA_PRODUCT_BUNDLE_IDENTIFIER': '',
        'XCTestConfigurationFilePath':'',
        'XCODE_DBG_XPC_EXCLUSIONS': 'com.apple.dt.xctestSymbolicator',
        'SWIFTUI_VIEW_DEBUG':287,
        'XCTestBundlePath': f"{app_info['Path']}/PlugIns/{target_name}.xctest",
         '__XCODE_BUILT_PRODUCTS_DIR_PATHS': '/tmp/derivedDataPath/Build/Products/Release-iphoneos',
         '__XPC_DYLD_FRAMEWORK_PATH': '/tmp/derivedDataPath/Build/Products/Release-iphoneos',
         '__XPC_DYLD_LIBRARY_PATH': '/tmp/derivedDataPath/Build/Products/Release-iphoneos',
        'MJPEG_SERVER_PORT': '',
        'USE_PORT': '',
        # maybe no needed
        'LLVM_PROFILE_FILE': app_container + "/tmp/%p.profraw", # %p means pid
    } # yapf: disable

你修改_device.py代码中env 看看是否还掉线

你自己试了有用？