alibaba/tidevice #145: 请文支持查看Documents有什么条件？我试了下我这边的测试app报错

sadmenmen, 2021-09-18:
tidevice fsync -B com.xiaomi.mihome ls /Documents/
Traceback (most recent call last):
  File "c:\python37\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\python37\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\python37\Scripts\tidevice.exe\__main__.py", line 7, in <module>
  File "c:\python37\lib\site-packages\tidevice\__main__.py", line 743, in main
    actions[args.subparser](args)
  File "c:\python37\lib\site-packages\tidevice\__main__.py", line 413, in cmd_fsync
    pprint(sync.listdir(arg0))
  File "c:\python37\lib\site-packages\tidevice\_sync.py", line 110, in listdir
    pkg = self._recv()
  File "c:\python37\lib\site-packages\tidevice\_sync.py", line 76, in _recv
    status_data = self.recvall(plist_size) # Discard plist xml-content
  File "c:\python37\lib\site-packages\tidevice\_safe_socket.py", line 61, in recvall
    raise MuxError("socket connection broken")
tidevice.exceptions.MuxError: socket connection broken

sadmenmen, 2021-09-18:
tidevice fsync -B com.xiaomi.mihome ls /Documents/

sadmenmen, 2021-09-18:
IOS 12.5.4
tidevice  0.5.0 
查看米家的文件提示中断，我看了下多看阅读是可以访问，但是目录为空。是因为米家没开放沙盒权限吗

liuqiao1995, 2021-11-22:
我也是报这个错

codeskyblue, 2022-04-01:
首先得app同意你才能看

closed 2022-04-01 by codeskyblue