alibaba/tidevice #170: MAC PRO i7 安装未报错，但是无法使用，tidevice -v 报TypeError: Expected maxsize to be an integer or None

Felix1029, 2021-12-14:
➜  ~ tidevice -v        
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.7/bin/tidevice", line 5, in <module>
    from tidevice.__main__ import main
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/tidevice/__init__.py", line 7, in <module>
    from ._device import BaseDevice as Device
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/tidevice/_device.py", line 35, in <module>
    from ._ipautil import IPAReader
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/tidevice/_ipautil.py", line 21, in <module>
    class IPAReader(zipfile.ZipFile):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/tidevice/_ipautil.py", line 56, in IPAReader
    def get_infoplist(self) -> dict:
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/functools.py", line 490, in lru_cache
    raise TypeError('Expected maxsize to be an integer or None')
TypeError: Expected maxsize to be an integer or None

Felix1029, 2021-12-15:
[image]
感觉是这个原因， 我是需要升级python 么

codeskyblue, 2022-03-23:
fixed

closed 2022-03-23 by codeskyblue