alibaba/tidevice #172: 运行后报tidevice.exceptions.MuxError: Launch failed

seminchen, 2021-12-18:
CFBundleExecutable: WebDriverAgentRunner-Runner
[I 211218 18:59:46 _device:841] AppContainer: /private/var/mobile/Containers/Data/Application/8621A7FE-15AB-4ADB-A763
[E 211218 18:59:46 _device:878] Launch failed: NSError(CODE:2 DOMAIN:com.apple.dt.deviceprocesscontrolservice INFO:{'NSLocalizedDescription': 'Request to launch '
                               'com.facebook.WebDriverAgentRunner123.xctrunner '
                               'failed.',
     'NSLocalizedFailureReason': 'The operation couldn’t be completed. (null) : '
                                 'Failed to launch process with bundle identifier '
                                 "'com.facebook.WebDriverAgentRunner123.xctrunner'.",
     'NSUnderlyingError': NSError(CODE:1 DOMAIN:FBSOpenApplicationErrorDomain INFO:{'BSErrorCodeDescription': 'Unspecified', 'NSLocalizedFailureReason': '(null)'})})
Traceback (most recent call last):
  File "D:\ProgramFiles\Python\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "D:\ProgramFiles\Python\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "D:\ProgramFiles\Python\lib\site-packages\tidevice\__main__.py", line 747, in <module>
    main()
  File "D:\ProgramFiles\Python\lib\site-packages\tidevice\__main__.py", line 742, in main
    actions[args.subparser](args)
  File "D:\ProgramFiles\Python\lib\site-packages\tidevice\__main__.py", line 237, in cmd_xctest
    d.xctest(args.bundle_id,
  File "D:\ProgramFiles\Python\lib\site-packages\tidevice\_device.py", line 1036, in xctest
    pid = self._launch_app_runner(bundle_id, session_identifier,
  File "D:\ProgramFiles\Python\lib\site-packages\tidevice\_device.py", line 879, in _launch_app_runner
    raise MuxError("Launch failed")
tidevice.exceptions.MuxError: Launch failed

sunlightzf, 2022-01-19:
hi,same issue.have you rosolve it?

sunlightzf, 2022-01-19:
同样的问题 你搞定了吗

daixiaoqin-1, 2022-03-21:
我好像每周一都会报这个错，然后重新安装wda又可以用一个星期

mynamecool, 2022-07-04:
同样这个问题 求解决方案

engzegngi-dd, 2022-07-11:
应该是证书过期的问题？可以尝试下重新构建并安装WDA。
这个报错内容类似：https://github.com/alibaba/taobao-iphone-device/issues/133

seminchen, 2022-07-15:
与WDA驱动有关，重新安装可以解决，就是失效频率稍高，证书是99$一年有效期的