alibaba/tidevice #173: 运行后报tidevice.exceptions.MuxError: Launch failed

seminchen, 2021-12-18:
CFBundleExecutable: WebDriverAgentRunner-Runner
[I 211218 18:59:46 _device:841] AppContainer: /private/var/mobile/Containers/Data/Application/8621A7FE-15AB-4ADB-A763
[E 211218 18:59:46 _device:878] Launch failed: NSError(CODE:2 DOMAIN:com.apple.dt.deviceprocesscontrolservice INFO:{'NSLocalizedDescription': 'Request to launch '
                               'com.facebook.WebDriverAgentRunner123.xctrunner '
                               'failed.',
     'NSLocalizedFailureReason': 'The operation couldn’t be completed. (null) : '
                                 'Failed to launch process with bundle identifier '
                                 "'com.facebook.WebDriverAgentRunner123.xctrunner'.",
     'NSUnderlyingError': NSError(CODE:1 DOMAIN:FBSOpenApplicationErrorDomain INFO:{'BSErrorCodeDescription': 'Unspecified', 'NSLocalizedFailureReason': '(null)'})})
Traceback (most recent call last):
  File "D:\ProgramFiles\Python\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "D:\ProgramFiles\Python\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "D:\ProgramFiles\Python\lib\site-packages\tidevice\__main__.py", line 747, in <module>
    main()
  File "D:\ProgramFiles\Python\lib\site-packages\tidevice\__main__.py", line 742, in main
    actions[args.subparser](args)
  File "D:\ProgramFiles\Python\lib\site-packages\tidevice\__main__.py", line 237, in cmd_xctest
    d.xctest(args.bundle_id,
  File "D:\ProgramFiles\Python\lib\site-packages\tidevice\_device.py", line 1036, in xctest
    pid = self._launch_app_runner(bundle_id, session_identifier,
  File "D:\ProgramFiles\Python\lib\site-packages\tidevice\_device.py", line 879, in _launch_app_runner
    raise MuxError("Launch failed")
tidevice.exceptions.MuxError: Launch failed

seminchen, 2022-01-01:
submit duplicate

closed 2022-01-01 by seminchen

mynamecool, 2022-07-01:
你好，请问这个问题怎么解决呢？ 我用的最新版的tidevice 还是存在这个问题