alibaba/tidevice #184: Is there any plan for stream video using this utility

bikeebh, 2022-01-08:
@codeskyblue @nanoscopic is there any plan for stream Video using tidevice 

if we can take screenshot using tidevice using same can we do Video streaming?

codeskyblue, 2022-04-01:
```
import tidevice
d = tidevice.Device()
for im in d.iter_screenshot():
    print(im.size)
```

The speed is not very fast, but might be enough to generate video

krishtoautomate, 2022-04-01:
Better to use avfoundation

nanoscopic, 2022-04-01:
Screenshots through lockdown are too slow to be useful as a video feed.

avfoundation works but tends to go down after a few days of continual use and then be finicky to get it to restart. If you are willing to reset avfoundation daily it works well.

Frames via WDA / CFA work well.

For the highest speed / quality Apple supported method, I'd recommend using an upload broadcast extension. They are just a bit tricky to start programmatically since Apple doesn't provide a direct way to do so.

krishtoautomate, 2022-04-01:
Avfoundation to websocket apis might help to start it, buffer raw data and stop connection on end

nanoscopic, 2022-04-01:
The problem with avfoundation is that the service on the phone that makes it work tends to crash / freeze. Also sometimes it simply refuses to start streaming.

To reliably start you need to first try to start it, and if that doesn't work kill the process responsible for it on the iOS device. If it still fails in worst case ( this does happen ) the phone may need to be rebooted.

The avfoundation method just isn't stable, as Apple has not maintained it. There continues to be no "official" documentation on it, only mention of it at the one conference. Basically a random employee hacked it in and they haven't bothered to make it reliable.