alibaba/tidevice #189: 不同机型下CPU同一场景下采集值相差较大，iPhone11-cpu 均值1%，iPhone12-cpu 均值30%

JayGitH, 2022-01-21:
不同机型下CPU同一场景下采集值相差较大，iPhone11-cpu 均值1%，iPhone12-cpu 均值30%
型号：iPhone11 pro iOS 13
型号：iPhone12 pro iOS 14.2
请问老师这个如何是何原因的，我要如何排查及解决？

JayGitH, 2022-01-24:
@codeskyblue @vdavdev 老师 帮看下这个问题呢 谢谢

JayGitH, 2022-01-26:
iPhone iOS 13.4.1
instruments和tidevice采集的相差很大
初步排查：iOS14的结果 tidevice和instruments的结果相近
[image]
[image]

lycfr, 2022-02-18:
14系统以下的需要 * 40 就可以了

JayGitH, 2022-02-18:
> 14系统以下的需要 * 40 就可以了

@lycfr 谢谢老师的解决方案。方便问下老师，通过什么方式得到解决这个问题的？我学习下，后续也给该项目添砖加瓦下。

lycfr, 2022-02-24:
@JayGitH 作者说的。之前在群里面有问过，回复的是 *40就可以了