alibaba/tidevice #192: tidevice list指令后卡住

A-Cepheus, 2022-01-28:
`tidevice list`指令后卡住
强行终止后发现卡在socket连接的过程
`List of apple devices attached
^CTraceback (most recent call last):
  File "/usr/local/bin/tidevice", line 8, in <module>
    sys.exit(main())
  File "/home/miyu/.local/lib/python3.8/site-packages/tidevice/__main__.py", line 766, in main
    actions[args.subparser](args)
  File "/home/miyu/.local/lib/python3.8/site-packages/tidevice/__main__.py", line 79, in cmd_list
    for dinfo in um.device_list():
  File "/home/miyu/.local/lib/python3.8/site-packages/tidevice/_usbmux.py", line 76, in device_list
    data = self.send_recv(payload)
  File "/home/miyu/.local/lib/python3.8/site-packages/tidevice/_usbmux.py", line 47, in send_recv
    with self.create_connection() as s:
  File "/home/miyu/.local/lib/python3.8/site-packages/tidevice/_usbmux.py", line 44, in create_connection
    return PlistSocket(self.__address, self._next_tag())
  File "/home/miyu/.local/lib/python3.8/site-packages/tidevice/_safe_socket.py", line 112, in __init__
    super().__init__(addr)
  File "/home/miyu/.local/lib/python3.8/site-packages/tidevice/_safe_socket.py", line 58, in __init__
    self._sock.connect(addr)
KeyboardInterrupt`

ruomubingfeng, 2022-02-14:
你下载安装一个itools或者爱思助手，应该就可以解决问题

A-Cepheus, 2022-02-14:
> 你下载安装一个itools或者爱思助手，应该就可以解决问题

问题已解决 手动启动usbmuxd可以查看相关输出 解决报错即可

closed 2022-02-14 by A-Cepheus