alibaba/tidevice #193: 文件操作可以支持读取ios的Library目录吗

mp55, 2022-02-08:
用爱思助手可以拉取Library里的目录，
[image]

tidevice fsync -B com.test.xxx pull Library/xx.ini
会提示
Traceback (most recent call last):
  File "c:\users\1\appdata\local\programs\python\python37\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\1\appdata\local\programs\python\python37\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\1\AppData\Local\Programs\Python\Python37\Scripts\tidevice.exe\__main__.py", line 7, in <module>
  File "c:\users\1\appdata\local\programs\python\python37\lib\site-packages\tidevice\__main__.py", line 742, in main
    actions[args.subparser](args)
  File "c:\users\1\appdata\local\programs\python\python37\lib\site-packages\tidevice\__main__.py", line 425, in cmd_fsync
    data = sync.pull_content(arg0)
  File "c:\users\1\appdata\local\programs\python\python37\lib\site-packages\tidevice\_sync.py", line 317, in pull_content
    for chunk in self.iter_content(path):
  File "c:\users\1\appdata\local\programs\python\python37\lib\site-packages\tidevice\_sync.py", line 297, in iter_content
    info = self.stat(path)
  File "c:\users\1\appdata\local\programs\python\python37\lib\site-packages\tidevice\_sync.py", line 172, in stat
    AFCStatus(pkg.status)))
tidevice.exceptions.MuxError: stat Library/xx.ini - AFCStatus.PERM_DENIED

codeskyblue, 2022-03-17:
试试这个命令
`tidevice fsync -B com.test.xxx ls /Library`

hushhw, 2022-08-18:
> 试试这个命令 `tidevice fsync -B com.test.xxx ls /Library`

ls 没有输出，第三方的app都无法获取沙盒文件吗