alibaba/tidevice #198: 这个错误麻烦处理下

pxrgod, 2022-03-17:
Traceback (most recent call last):
  File "/opt/homebrew/Cellar/python@3.9/3.9.5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/opt/homebrew/Cellar/python@3.9/3.9.5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/hspcadmin/Library/Python/3.9/lib/python/site-packages/tidevice/_perf.py", line 251, in append_data
    for _type, data in idata:
  File "/Users/hspcadmin/Library/Python/3.9/lib/python/site-packages/tidevice/_perf.py", line 245, in iter_network_flow
    "upFlow": nstat['tx.bytes'] / 1024
TypeError: unsupported operand type(s) for /: 'NSNull' and 'int'

codeskyblue, 2022-03-17:
NSNull啊，nstat竟然还有NSNull数据

codeskyblue, 2022-03-17:
fixed in 0.6.7

commit 08b87340 (2022-03-17): fix #198
closed 2022-03-17 by codeskyblue