alibaba/tidevice #204: 执行xctest报错超时退出

jongyu, 2022-04-12:
环境：Windows 10
设备：iPhone 12（15.4）
Python版本：3.9.1
tidevice版本：0.7.6
执行命令：`tidevice xctest -B com.xxx.xctrunner`
报错内容
```bash
C:\Users\user>tidevice xctest -B com.xxx.xctrunner
[I 220412 10:54:58 _device:977] ProductVersion: 15.4
[I 220412 10:54:58 _device:824] CFBundleExecutable: FastbotRunner-Runner
[I 220412 10:54:59 _device:855] AppContainer: /private/var/mobile/Containers/Data/Application/EDB662B5-A0FF-40CF-8156-6C14B9F66D2F
[I 220412 10:54:59 _device:895] Launch 'com.xxx.xctrunner' pid: 45661
[I 220412 10:54:59 _device:1028] Test runner ready detected
[I 220412 10:54:59 _device:1020] Start execute test plan with IDE version: 29
[E 220412 10:56:59 _instruments:573] drain error
    Traceback (most recent call last):
      File "c:\users\user\appdata\local\programs\python\python39\lib\site-packages\tidevice\_instruments.py", line 563, in _drain
        self._drain_single_message()
      File "c:\users\user\appdata\local\programs\python\python39\lib\site-packages\tidevice\_instruments.py", line 586, in _drain_single_message
        mheader, payload = self.recv_dtx_message()
      File "c:\users\user\appdata\local\programs\python\python39\lib\site-packages\tidevice\_instruments.py", line 480, in recv_dtx_message
        data = self.psock.recvall(0x20)
      File "c:\users\user\appdata\local\programs\python\python39\lib\site-packages\tidevice\_safe_socket.py", line 88, in recvall
        chunk = self._sock.recv(size - len(buf))
      File "c:\users\user\appdata\local\programs\python\python39\lib\ssl.py", line 1226, in recv
        return self.read(buflen)
      File "c:\users\user\appdata\local\programs\python\python39\lib\ssl.py", line 1101, in read
        return self._sslobj.read(len)
    socket.timeout: The read operation timed out
[I 220412 10:56:59 _device:1100] xctrunner quited
```

ferdyzhang, 2022-04-19:
我也遇到了同样的问题，使用tidevice。但在xcode上比较稳定，不会timeout.还不知道怎么解决

wangfeng3-1402, 2022-04-19:
[image]
嗯，一样的问题

ferdyzhang, 2022-04-21:
> 我也遇到了同样的问题，使用tidevice。但在xcode上比较稳定，不会timeout.还不知道怎么解决
> 
> chunk = self._sock.recv(size - len(buf)) File "/usr/lib/python3.9/ssl.py", line 1226, in recv return self.read(buflen) File "/usr/lib/python3.9/ssl.py", line 1101, in read return self._sslobj.read(len) socket.timeout: The read operation timed out

我现在是写了一个shell脚本，定期检查这个tidevice进程是否alive,否则就把他拉起来。这个方法比较笨😂

tinysheepyang, 2022-05-07:
同遇到此问题，有解决方法的大佬吗？

tinysheepyang, 2022-05-07:
建议用0.7.4版本，0.7.6 0.7.9版本会有这样的问题

codeskyblue, 2022-05-09:
这个问题在0.7.11版本已修复。
感谢反馈。不过建议用更新的版本0.7.12，这个版本还修复了过去的一个mheader可能为None的问题

codeskyblue, 2022-05-09:
暂时关闭，如果有问题再reopen

closed 2022-05-09 by codeskyblue

xiaohengdai, 2022-08-22:
大佬，0.9.7会遇到[E 220822 14:27:40 _instruments:594] drain error， Traceback (most recent call last):
      File "d:\pdd\code\iosuiautomation\venv\lib\site-packages\tidevice\_instruments.py", line 580, in _drain
        self._drain_single_message()
      File "d:\pdd\code\iosuiautomation\venv\lib\site-packages\tidevice\_instruments.py", line 607, in _drain_single_message
        mheader, payload = self.recv_dtx_message()
      File "d:\pdd\code\iosuiautomation\venv\lib\site-packages\tidevice\_instruments.py", line 513, in recv_dtx_message
        message_id = self.recv_part_dtx_message()
      File "d:\pdd\code\iosuiautomation\venv\lib\site-packages\tidevice\_instruments.py", line 409, in recv_part_dtx_message
        data = self.psock.recvall(0x20)
      File "d:\pdd\code\iosuiautomation\venv\lib\site-packages\tidevice\_safe_socket.py", line 111, in recvall
        chunk = self._sock.recv(size - len(buf))
      File "D:\软件\python\lib\ssl.py", line 994, in recv
        return self.read(buflen)
      File "D:\软件\python\lib\ssl.py", line 871, in read
        return self._sslobj.read(len, buffer)
      File "D:\软件\python\lib\ssl.py", line 633, in read
    ConnectionAbortedError: [WinError 10053] 你的主机中的软件中止了一个已建立的连接。
tidevice                0.9.7