alibaba/tidevice #216: crashreport导出Crash文件是会产生空文集

jongyu, 2022-05-11:
使用`tidevice crashreport`命令导出App崩溃文件时，有时文件内容为空，非必先，出现了好几次

codeskyblue, 2022-05-17:
crash内容是先调用一个服务，将crash内容复制过来，然后再执行的下载。
我猜为空的时候，应该是复制过来的内容本身就是空的。
可以先导出前等待一下，或者看到空的就再次尝试一下。

dddzname, 2022-06-15:
是直接执行tidevice crashreport就可以吗？执行了提示错误信息
tidevice crashreport
usage: tidevice [-h] [-v] [-u UDID] [--socket SOCKET]
                {version,list,info,date,sysinfo,appinfo,applist,battery,screenshot,install,uninstall,reboot,shutdown,parse,watch,wait-for-device,launch,kill,ps,relay,xctest,wdaproxy,syslog,fsync,dumpfps,developer,pair,unpair,perf,set-assistive-touch,savesslfile,test}
                ...
tidevice: error: argument subparser: invalid choice: 'crashreport' (choose from 'version', 'list', 'info', 'date', 'sysinfo', 'appinfo', 'applist', 'battery', 'screenshot', 'install', 'uninstall', 'reboot', 'shutdown', 'parse', 'watch', 'wait-for-device', 'launch', 'kill', 'ps', 'relay', 'xctest', 'wdaproxy', 'syslog', 'fsync', 'dumpfps', 'developer', 'pair', 'unpair', 'perf', 'set-assistive-touch', 'savesslfile', 'test')

lycfr, 2022-06-15:
@dddzname 版本太低了吧，升级到最新版本应该就可以了