alibaba/tidevice #223: tidevice wdaproxy 无法挂在14.6的DeveloperDiskImage

pkm1118, 2022-05-30:
我iPhone14.6的设备在使用tidevice wdaproxy 命令时，报如下错误有人知道怎么解决吗 （我已经将下载的14.6解压到/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/DeviceSupport/ 目录中了）

[I 220530 14:44:48 _wdaproxy:125] [d7d15615997597abff977fb66843ff1996c644b8] WDA check every 30.0 seconds
[D 220530 14:44:48 _wdaproxy:131] [d7d15615997597abff977fb66843ff1996c644b8] launch WDA
[I 220530 14:44:49 _device:969] BundleID: com.facebook.d7d15615997597abff977fb66843ff1996c644b8.WebDriverAgentRunner.xctrunner
[I 220530 14:44:49 _device:985] ProductVersion: 14.6
[I 220530 14:44:49 _device:986] DeviceIdentifier: d7d15615997597abff977fb66843ff1996c644b8
[I 220530 14:44:49 _device:607] ProductVersion: 14.6
[I 220530 14:44:49 _imagemounter:157] Pushing DeveloperDiskImage.dmg
[I 220530 14:44:50 _imagemounter:170] Push complete
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/tidevice/_device.py", line 492, in start_service
    return self._unsafe_start_service(name)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/tidevice/_device.py", line 508, in _unsafe_start_service
    raise MuxServiceError(error)
tidevice.exceptions.MuxServiceError: InvalidService

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/tidevice/__main__.py", line 871, in <module>
    main()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/tidevice/__main__.py", line 866, in main
    actions[args.subparser](args)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/tidevice/__main__.py", line 259, in cmd_xctest
    d.xctest(args.bundle_id,
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/tidevice/_device.py", line 970, in xctest
    return self.xcuitest(bundle_id, target_bundle_id=target_bundle_id, logger=logger, env=env)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/tidevice/_device.py", line 996, in xcuitest
    x1 = self._connect_testmanagerd_lockdown()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/tidevice/_device.py", line 791, in _connect_testmanagerd_lockdown
    conn = self.start_service(
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/tidevice/_device.py", line 494, in start_service
    self.mount_developer_image()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/tidevice/_device.py", line 668, in mount_developer_image
    self.imagemounter.mount(image_path, signature_path)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/tidevice/_imagemounter.py", line 139, in mount
    return self.mount_fileobj(image_reader, image_size, signature_content)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/tidevice/_imagemounter.py", line 182, in mount_fileobj
    self._check_error(ret)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/tidevice/_imagemounter.py", line 123, in _check_error
    raise MuxError(ret['Error'])
tidevice.exceptions.MuxError: ImageMountFailed
[E 220530 14:44:50 _wdaproxy:150] [d7d15615997597abff977fb66843ff1996c644b8] wda started failed
[D 220530 14:44:50 _wdaproxy:131] [d7d15615997597abff977fb66843ff1996c644b8] launch WDA

codeskyblue, 2022-06-13:
重启一下手机说不定就好了

pkm1118, 2022-06-13:
已经解决了，因为手机中的/Deveolper目录为非空目录，导致DeveolperImage一直挂载失败，只要将手机中的/Deveolper 清空即可以成功挂载

closed 2022-07-26 by codeskyblue

li8911255, 2022-09-15:
> 已经解决了，因为手机中的/Deveolper目录为非空目录，导致DeveolperImage一直挂载失败，只要将手机中的/Deveolper 清空即可以成功挂载

请问用什么方式可以清理这个目录？我也遇到同样的问题了

sumuzhe123, 2023-01-30:
也是遇到同样问题，请问用什么方式可以清理这个/Deveolper目录？

sumuzhe123, 2023-01-30:
> 已经解决了，因为手机中的/Deveolper目录为非空目录，导致DeveolperImage一直挂载失败，只要将手机中的/Deveolper 清空即可以成功挂载

这个要怎么删除手机中的/Deveolper目录的内容呢？是需要越狱吗？

pkm1118, 2023-02-01:
> > 已经解决了，因为手机中的/Developer目录为非空间目录，导致DeveolperImage一直接挂载丢失，只要将手机中的/Developer清空即可成功挂载
> 
> 这个要怎么删除手机中的/Developer目录的内容呢？是需要越狱吗？

是的，需要越狱后连接到手机进行删除

sumuzhe123, 2023-02-01:
> 

有没有其他不越狱的解决办法呢？

sumuzhe123, 2023-02-01:
> > > 已经解决了，因为手机中的/Developer目录为非空间目录，导致DeveolperImage一直接挂载丢失，只要将手机中的/Developer清空即可成功挂载
> > 
> > 
> > 这个要怎么删除手机中的/Developer目录的内容呢？是需要越狱吗？
> 
> 是的，需要越狱后连接到手机进行删除

有没有其他不越狱的解决办法呢？