alibaba/tidevice #232: win10下 每30秒一次started fail，最后程序退出

j188482367, 2022-06-24:
[I 220624 14:50:49 _wdaproxy:125] [c61e70c1026767f5e1b73e28be4baee61e827beb] WDA check every 30.0 seconds                                                       [D 220624 14:50:49 _wdaproxy:131] [c61e70c1026767f5e1b73e28be4baee61e827beb] launch WDA                                                                         [I 220624 14:50:50 _device:963] BundleID: com.facebook.WebDriverAgentRunner.xctrunner                                                                           [I 220624 14:50:50 _device:979] ProductVersion: 14.6                                                                                                            [I 220624 14:50:50 _device:980] DeviceIdentifier: c61e70c1026767f5e1b73e28be4baee61e827beb                                                                      [I 220624 14:50:50 _device:816] SignIdentity: 'Apple Development: junhong liu (GC2G6B3EW2)'                                                                     [I 220624 14:50:50 _device:822] CFBundleExecutable: WebDriverAgentRunner-Runner                                                                                 [I 220624 14:50:51 _device:853] AppContainer: /private/var/mobile/Containers/Data/Application/43018006-1958-4868-93A9-270A6BB6F8EF                              [I 220624 14:50:51 _device:893] Launch 'com.facebook.WebDriverAgentRunner.xctrunner' pid: 2636                                                                  [I 220624 14:50:51 _device:1029] Test runner ready detected                                                                                                     [I 220624 14:50:51 _device:1021] Start execute test plan with IDE version: 29                                                                                   [I 220624 14:50:52 _device:929] WebDriverAgent start successfully                                                                                               [E 220624 14:51:20 _wdaproxy:150] [c61e70c1026767f5e1b73e28be4baee61e827beb] wda started failed                                                                 [D 220624 14:51:20 _wdaproxy:131] [c61e70c1026767f5e1b73e28be4baee61e827beb] launch WDA                                                                         [I 220624 14:51:20 _device:963] BundleID: com.facebook.WebDriverAgentRunner.xctrunner                                                                           [I 220624 14:51:20 _device:979] ProductVersion: 14.6                                                                                                            [I 220624 14:51:20 _device:980] DeviceIdentifier: c61e70c1026767f5e1b73e28be4baee61e827beb                                                                      [I 220624 14:51:20 _device:816] SignIdentity: 'Apple Development: junhong liu (GC2G6B3EW2)'                                                                     [I 220624 14:51:20 _device:822] CFBundleExecutable: WebDriverAgentRunner-Runner                                                                                 [I 220624 14:51:21 _device:853] AppContainer: /private/var/mobile/Containers/Data/Application/43018006-1958-4868-93A9-270A6BB6F8EF                              [I 220624 14:51:21 _device:893] Launch 'com.facebook.WebDriverAgentRunner.xctrunner' pid: 2640                                                                  [I 220624 14:51:21 _device:1029] Test runner ready detected                                                                                                     [I 220624 14:51:21 _device:1021] Start execute test plan with IDE version: 29                                                                                   [I 220624 14:51:22 _device:929] WebDriverAgent start successfully                                                                                               [E 220624 14:51:50 _wdaproxy:150] [c61e70c1026767f5e1b73e28be4baee61e827beb] wda started failed                                                                 [D 220624 14:51:50 _wdaproxy:131] [c61e70c1026767f5e1b73e28be4baee61e827beb] launch WDA                                                                         [I 220624 14:51:50 _device:963] BundleID: com.facebook.WebDriverAgentRunner.xctrunner                                                                           [I 220624 14:51:50 _device:979] ProductVersion: 14.6                                                                                                            [I 220624 14:51:50 _device:980] DeviceIdentifier: c61e70c1026767f5e1b73e28be4baee61e827beb                                                                      [I 220624 14:51:51 _device:816] SignIdentity: 'Apple Development: junhong liu (GC2G6B3EW2)'                                                                     [I 220624 14:51:51 _device:822] CFBundleExecutable: WebDriverAgentRunner-Runner                                                                                 [I 220624 14:51:51 _device:853] AppContainer: /private/var/mobile/Containers/Data/Application/43018006-1958-4868-93A9-270A6BB6F8EF                              [I 220624 14:51:51 _device:893] Launch 'com.facebook.WebDriverAgentRunner.xctrunner' pid: 2644                                                                  [I 220624 14:51:51 _device:1029] Test runner ready detected                                                                                                     [I 220624 14:51:51 _device:1021] Start execute test plan with IDE version: 29                                                                                   [I 220624 14:51:52 _device:929] WebDriverAgent start successfully                                                                                               [E 220624 14:52:20 _wdaproxy:150] [c61e70c1026767f5e1b73e28be4baee61e827beb] wda started failed                                                                 [D 220624 14:52:20 _wdaproxy:131] [c61e70c1026767f5e1b73e28be4baee61e827beb] launch WDA                                                                         [I 220624 14:52:21 _device:963] BundleID: com.facebook.WebDriverAgentRunner.xctrunner                                                                           [I 220624 14:52:21 _device:979] ProductVersion: 14.6                                                                                                            [I 220624 14:52:21 _device:980] DeviceIdentifier: c61e70c1026767f5e1b73e28be4baee61e827beb                                                                      [I 220624 14:52:21 _device:816] SignIdentity: 'Apple Development: junhong liu (GC2G6B3EW2)'                                                                     [I 220624 14:52:21 _device:822] CFBundleExecutable: WebDriverAgentRunner-Runner                                                                                 [I 220624 14:52:21 _device:853] AppContainer: /private/var/mobile/Containers/Data/Application/43018006-1958-4868-93A9-270A6BB6F8EF                              [I 220624 14:52:21 _device:893] Launch 'com.facebook.WebDriverAgentRunner.xctrunner' pid: 2648                                                                  [I 220624 14:52:21 _device:1029] Test runner ready detected                                                                                                     [I 220624 14:52:21 _device:1021] Start execute test plan with IDE version: 29                                                                                   [I 220624 14:52:22 _device:929] WebDriverAgent start successfully                                                                                               [E 220624 14:52:50 _wdaproxy:150] [c61e70c1026767f5e1b73e28be4baee61e827beb] wda started failed                                                                 [D 220624 14:52:50 _wdaproxy:131] [c61e70c1026767f5e1b73e28be4baee61e827beb] launch WDA                                                                         [I 220624 14:52:51 _device:963] BundleID: com.facebook.WebDriverAgentRunner.xctrunner                                                                           [I 220624 14:52:51 _device:979] ProductVersion: 14.6                                                                                                            [I 220624 14:52:51 _device:980] DeviceIdentifier: c61e70c1026767f5e1b73e28be4baee61e827beb                                                                      [I 220624 14:52:51 _ [... cut ...]                     [I 220624 14:52:51 _device:1021] Start execute test plan with IDE version: 29                                                                                   [I 220624 14:52:52 _device:929] WebDriverAgent start successfully                                                                                               [E 220624 14:53:20 _wdaproxy:150] [c61e70c1026767f5e1b73e28be4baee61e827beb] wda started failed  

Mac 下面正常，两台win10  都是21H2 ，其中一台会出现30秒一次started fail ，第4次就自动结束了

xiaoMGitHub, 2022-06-27:
我也遇到这情况，用这两条命令启动就没问题
tidevice xctest -B com.facebook.wda.WebDriverAgent.Runner
tidevice relay 8100 8100

j188482367, 2022-06-27:
> Runner

不玩window了，现在用m1 mini 一带20

codeskyblue, 2022-07-26:
fixed in 0.9.7

closed 2022-07-26 by codeskyblue