alibaba/tidevice #239: It doesn't work on the secondary process

LRF520, 2022-07-17:
It just works on the main process, it doesn't work on the secondary process(subprocess), just like Wechat applet,maybe all of the data about the secondary process is wrong . so many apps have the secondary process now, it's better to support secondary process, it's better to show the total data of phone,so that we can avoid fatal mistakes.

codeskyblue, 2022-07-19:
em... Can you specify you problem you meet?

LRF520, 2022-08-19:
> em... Can you specify you problem you meet?

Don't just monitor the PID of the package name and package name, because there are many sub process names and sub process PIDs under the package name. Similar to such names, they have their own independent PIDs. They will independently consume CPU, memory, network, GPU and FPS resources. In many cases, the resource consumption of the sub process is much larger than that of the main process. Sometimes, the main process only works in the background, The main process does not consume a significant proportion of resources, so it is meaningless to only monitor the main process. If you want to get the real performance data of an app, you must count the performance data of all PIDs of the app, and summarize and sum all the data. There is a very important point here, that is, the PID of these sub processes is constantly changing, while the PID of the main process is always fixed. Only the performance data of the main process can not reflect the real situation. This statistical mode is also used in your Android performance testing tools, so the performance data statistical model on Android is also wrong. I have carefully analyzed both tools.

For example, wechat applet is an independent sub process of wechat, and they are constantly changing. You can try it (this is just a typical example):

com.tencent.mm/com.tencent.mm.plugin.appbrand.ui.AppBrandUI