alibaba/tidevice #240: wdaproxy命令无法启动WDA服务

jongyu, 2022-07-25:
- tidevice版本`0.9.6`
- 系统
```
ProductName:	macOS
ProductVersion:	12.4
BuildVersion:	21F79
```

- 执行命令`tidevice wdaproxy -B com.facebook.WebDriverAgentRunner.xctrunner`及报错信息
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/tidevice/_wdaproxy.py", line 237, in _wrapped_func
    return self._func(*args, **self._kwargs)
  File "/usr/local/lib/python3.9/site-packages/tidevice/_wdaproxy.py", line 127, in _keep_wda_running
    self.logger.info("WDA check every %.1f seconds", check_interval)
AttributeError: 'functools._lru_cache_wrapper' object has no attribute 'info'
```

- 设备信息
```
MarketName:       iPhone 12 Pro
DeviceName:       XXXX
ProductVersion:   14.4
ProductType:      iPhone13,3
```

jongyu, 2022-07-25:
执行`tidevice xctest`命令是正常的

liufliufliuf, 2022-07-26:
我也遇到了同样的问题

```
$tidevice wdaproxy -B com.meritco.WebDriverAgentRunner.xctrunner --port 8100
Traceback (most recent call last):
  File "/Users/liuf/opt/anaconda3/lib/python3.9/site-packages/tidevice/_wdaproxy.py", line 237, in _wrapped_func
    return self._func(*args, **self._kwargs)
  File "/Users/liuf/opt/anaconda3/lib/python3.9/site-packages/tidevice/_wdaproxy.py", line 127, in _keep_wda_running
    self.logger.info("WDA check every %.1f seconds", check_interval)
AttributeError: 'functools._lru_cache_wrapper' object has no attribute 'info'
```

codeskyblue, 2022-07-26:
Python版本多少呀？

codeskyblue, 2022-07-26:
fixed in 0.9.7.

closed 2022-07-26 by codeskyblue