alibaba/tidevice #243: ios下tidevice打开多个app后会报错, 重启后才能恢复

liufliufliuf, 2022-08-01:
我的测试任务需要按照顺序打开和关闭200个app

在ios上使用airtest的start_app()会报错, 改为直接用subprocess调用"tidevice lauch ***", 大约跑了几十个app后就会报以下错误, 只有重启iphone后才能再打开app

`    def start_app(self, package):
        if self.plat == "android":
            self.device.start_app(package=package)
        else:
            subprocess.check_output(f"tidevice launch {package}", shell=True)

    def stop_app(self, package):
        if self.plat == "android":
            self.device.stop_app(package=package)
        else:
            subprocess.check_output(f"tidevice kill {package}", shell=True)
`

`start app_id: 203, name: 海底捞, deviceName: Jicheng的iPhonese2
Traceback (most recent call last):
  File "/Users/liuf/opt/anaconda3/bin/tidevice", line 8, in <module>
    sys.exit(main())
  File "/Users/liuf/opt/anaconda3/lib/python3.9/site-packages/tidevice/__main__.py", line 951, in main
    actions[args.subparser](args)
  File "/Users/liuf/opt/anaconda3/lib/python3.9/site-packages/tidevice/__main__.py", line 362, in cmd_kill
    pid = d.app_kill(args.name)
  File "/Users/liuf/opt/anaconda3/lib/python3.9/site-packages/tidevice/_device.py", line 690, in app_kill
    return self.app_stop(*args, **kwargs)
  File "/Users/liuf/opt/anaconda3/lib/python3.9/site-packages/tidevice/_device.py", line 682, in app_stop
    for p in ps:
  File "/Users/liuf/opt/anaconda3/lib/python3.9/site-packages/tidevice/_instruments.py", line 737, in app_process_list
    processes = self.app_running_processes()
  File "/Users/liuf/opt/anaconda3/lib/python3.9/site-packages/tidevice/_instruments.py", line 708, in app_running_processes
    retobj = self.call_message(identifier, "runningProcesses")
  File "/Users/liuf/opt/anaconda3/lib/python3.9/site-packages/tidevice/_instruments.py", line 357, in call_message
    channel = self.make_channel(channel)
  File "/Users/liuf/opt/anaconda3/lib/python3.9/site-packages/tidevice/_instruments.py", line 318, in make_channel
    result = self.call_message(0, '_requestChannelWithCode:identifier:', aux)
  File "/Users/liuf/opt/anaconda3/lib/python3.9/site-packages/tidevice/_instruments.py", line 359, in call_message
    _id = self.send_dtx_message(channel,
  File "/Users/liuf/opt/anaconda3/lib/python3.9/site-packages/tidevice/_instruments.py", line 398, in send_dtx_message
    self.psock.sendall(data)
  File "/Users/liuf/opt/anaconda3/lib/python3.9/site-packages/tidevice/_safe_socket.py", line 118, in sendall
    return self._sock.sendall(data)
  File "/Users/liuf/opt/anaconda3/lib/python3.9/ssl.py", line 1204, in sendall
    v = self.send(byte_view[count:])
  File "/Users/liuf/opt/anaconda3/lib/python3.9/ssl.py", line 1173, in send
    return self._sslobj.write(data)
ssl.SSLZeroReturnError: TLS/SSL connection has been closed (EOF) (_ssl.c:2483)
`

系统信息
macOS 12.5
iphone 15.6
python 3.9
tidevice 0.9.7

设备
iPhone SE (2nd generation)

Vantermonde, 2022-09-02:
同问，遇到这个问题之后有除了重启以外的解决方案吗

taozuia, 2022-09-24:
同问，遇到这个问题之后有除了重启以外的解决方案吗

lookdczar, 2022-12-13:
也是同样的问题，同一个app多次关闭、启动若干次后就会出现

codeskyblue, 2022-12-13:
升级一下看看好了没，版本0.9.13