alibaba/tidevice #249: 执行tidevice ps和tidevice kill com.xxxxx时，会出现以下报错：

screenbeam-Ricardo, 2022-08-19:
Traceback (most recent call last):
  File "E:\Python\Python38\lib\runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "E:\Python\Python38\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "E:\Python\Python38\Scripts\tidevice.exe\__main__.py", line 7, in <module>
  File "E:\Python\Python38\lib\site-packages\tidevice\__main__.py", line 951, in main
    actions[args.subparser](args)
  File "E:\Python\Python38\lib\site-packages\tidevice\__main__.py", line 362, in cmd_kill
    pid = d.app_kill(args.name)
  File "E:\Python\Python38\lib\site-packages\tidevice\_device.py", line 690, in app_kill
    return self.app_stop(*args, **kwargs)
  File "E:\Python\Python38\lib\site-packages\tidevice\_device.py", line 674, in app_stop
    with self.connect_instruments() as ts:
  File "E:\Python\Python38\lib\site-packages\tidevice\_device.py", line 792, in connect_instruments
    conn = self.start_service(
  File "E:\Python\Python38\lib\site-packages\tidevice\_device.py", line 483, in start_service
    return self._unsafe_start_service(name)
  File "E:\Python\Python38\lib\site-packages\tidevice\_device.py", line 522, in _unsafe_start_service
    conn = self.create_inner_connection(data['Port'], _ssl=_ssl, ssl_dial_only=ssl_dial_only)
  File "E:\Python\Python38\lib\site-packages\tidevice\_device.py", line 294, in create_inner_connection
    psock.switch_to_ssl(self.ssl_pemfile_path)
  File "E:\Python\Python38\lib\site-packages\tidevice\_safe_socket.py", line 142, in switch_to_ssl
    ssock = context.wrap_socket(self._sock, server_hostname="iphone.localhost")
  File "E:\Python\Python38\lib\ssl.py", line 500, in wrap_socket
    return self.sslsocket_class._create(
  File "E:\Python\Python38\lib\ssl.py", line 1040, in _create
    self.do_handshake()
  File "E:\Python\Python38\lib\ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
ConnectionAbortedError: [WinError 10053] 你的主机中的软件中止了一个已建立的连接。

测试环境：Windows 10
设备：ipad pro以及ipad 5
设备版本：15.2,15.4.1
tidevice版本:0.6.4

想请教下可能是什么原因导致的？

screenbeam-Ricardo, 2022-08-19:
重启手机后可以解决,但是又有了另一个问题，运行tidevice rebot后，电脑会无法识别这个设备,需要重新插拔数据线，请问下这个问题有解决办法么？

zhang25121, 2022-08-25:
运行tidevice reboot后，电脑会无法识别这个设备,需要重新插拔数据线
MAC也遇到同样问题，非必现，几率性的

codeskyblue, 2022-09-07:
@screenbeam-Ricardo 换个高版本的tidevice试试

我用0.9.11版本测试了启动应用设置100次，还没有遇到问题

```python
import tidevice

BUNDLE_ID = "com.apple.Preferences"

def main():

    t = tidevice.Device()
    for i in range(100):
        print(i, "times")
        t.app_start(BUNDLE_ID, kill_running=True)

if __name__ == "__main__":
    main()
```

screenbeam-Ricardo, 2022-09-07:
> @screenbeam-Ricardo 换个高版本的tidevice试试
>
> 我用0.9.11版本测试了启动应用设置100次，还没有遇到问题
>
> ```python
> import tidevice
>
> BUNDLE_ID = "com.apple.Preferences"
>
> def main():
>
>     t = tidevice.Device()
>     for i in [...]

> @screenbeam-Ricardo 换个高版本的tidevice试试
>
> 我用0.9.11版本测试了启动应用设置100次，还没有遇到问题
>
> ```python
> import tidevice
>
> BUNDLE_ID = "com.apple.Preferences"
>
> def main():
>
>     t = tidevice.Device()
>     for i in [...]

> @screenbeam-Ricardo 换个高版本的tidevice试试
>
> 我用0.9.11版本测试了启动应用设置100次，还没有遇到问题
>
> ```python
> import tidevice
>
> BUNDLE_ID = "com.apple.Preferences"
>
> def main():
>
>     t = tidevice.Device()
>     for i in [...]

> @screenbeam-Ricardo 换个高版本的tidevice试试
>
> 我用0.9.11版本测试了启动应用设置100次，还没有遇到问题
>
> ```python
> import tidevice
>
> BUNDLE_ID = "com.apple.Preferences"
>
> def main():
>
>     t = tidevice.Device()
>     for i in [...]

感谢回复，我前几天已经将tidevice升级到了0.9.7的版本，随即出现了新的问题:
(1)
11:09:36 Traceback (most recent call last):
11:09:36   File "D:\python\python38\lib\site-packages\tidevice\_instruments.py", line 566, in wait_reply
11:09:36     ret = self._reply_queues[message_id].get(timeout=timeout)
11:09:36   File "D:\python\python38\lib\queue.py", line 178, in get
11:09:36     raise Empty
11:09:36 _queue.Empty
11:09:36 
11:09:36 During handling of the above exception, another exception occurred:
11:09:36 
11:09:36 Traceback (most recent call last):
11:09:36   File "D:\python\python38\lib\runpy.py", line 193, in _run_module_as_main
11:09:36     return _run_code(code, main_globals, None,
11:09:36   File "D:\python\python38\lib\runpy.py", line 86, in _run_code
11:09:36     exec(code, run_globals)
11:09:36   File "D:\python\python38\Scripts\tidevice.EXE\__main__.py", line 7, in <module>
11:09:36   File "D:\python\python38\lib\site-packages\tidevice\__main__.py", line 951, in main
11:09:36     actions[args.subparser](args)
11:09:36   File "D:\python\python38\lib\site-packages\tidevice\__main__.py", line 268, in cmd_xctest
11:09:36     d.xctest(args.bundle_id,
11:09:36   File "D:\python\python38\lib\site-packages\tidevice\_device.py", line 963, in xctest
11:09:36     return self.xcuitest(bundle_id, target_bundle_id=target_bundle_id, logger=logger, env=env)
11:09:36   File "D:\python\python38\lib\site-packages\tidevice\_device.py", line 1050, in xcuitest
11:09:36     _, pid = self._launch_app_runner(bundle_id, session_identifier,
11:09:36   File "D:\python\python38\lib\site-packages\tidevice\_device.py", line 841, in _launch_app_runner
11:09:36     channel = conn.make_channel(InstrumentsService.ProcessControl)
11:09:36   File "D:\python\python38\lib\site-packages\tidevice\_instruments.py", line 318, in make_channel
11:09:36     result = self.call_message(0, '_requestChannelWithCode:identifier:', aux)
11:09:36   File "D:\python\python38\lib\site-packages\tidevice\_instruments.py", line 363, in call_message
11:09:36     return self.wait_reply(_id).result
11:09:36   File "D:\python\python38\lib\site-packages\tidevice\_instruments.py", line 571, in wait_reply
11:09:36     raise ServiceError("wait reply timeout")
11:09:36 tidevice.exceptions.ServiceError: wait reply timeout

使用tidevice ps，tidevice kill  xxxx都有时候会出现，我想会不会是因为笔记本电脑上通过Usb-Hub同时连接5个iOS设备，出现连接不稳定导致的?
我将出现这个现象的设备单独连到另外一个PC上也没有复现.

nicexp, 2023-03-29:
[image]
很简单的压测代码，启动杀死进程200多次的时候必现，inscrument的com.apple.instruments.server.services.deviceinfo服务直接没有响应，初步怀疑是每次执行创建的socket connection太多，反复跟服务建立连接导致 @codeskyblue

codeskyblue, 2023-03-30:
@nicexp 虽然是这么说，我特定每次用完之后就把连接都关掉了（最开始的版本没有关掉，很快就不行了）
最新的版本我使用retry在连接instruments服务的时候加了一个重试（重试次数3次，分别等待5s，8s，11s）

再试试看，我感觉现在好像稳定多了

另外kill应用也可以不用tidevice，直接用WDA的app停止，也可以。