alibaba/tidevice #251: exception happend when run tidevice fsync -B ${BundleId} ls /Documents/

DrxWeb, 2022-08-24:
### Problem discription:
[image]

Command **_tidevice fsync ls /DCIM/_**    works fine. But Error happend when  **_tidevice fsync -B ${BundleId} ls ./_** was excuted. 

### Environment info:
python --version
Python 2.7.17

tidevice -v
0.7.3

iOS system version
iPhone 6s         11.2.6            usb
iPhone 8 Plus   15.5              usb

is there any solution for solving the problem i met?

DrxWeb, 2022-08-24:
Same problem as 
#100 #124

codeskyblue, 2022-09-07:
需要应用 开启沙盒权限，具体参考 https://stackoverflow.com/questions/6029916/how-to-enable-file-sharing-for-my-app