alibaba/tidevice #254: tidevice kill "wda的包名"后tidevice ps中依旧存在

screenbeam-Ricardo, 2022-09-01:
如题，之前用的是tidevice-0.6.4，昨天更新到0.9.7,在跑用例的时候通过subprocess.Popen("tidevice kill '包名'")结束wda，重复10次都是无效的，想请教下这是什么原因导致的呢?或者有没有其他方法关闭设置上正在运行wda呢？

codeskyblue, 2022-09-07:
看看是不是包名没写对，我试了下看起来是可以的
```
tidevice kill com.facebook.WebDriverAgentRunner.xctrunner
```

screenbeam-Ricardo, 2022-09-08:
> 看看是不是包名没写对，我试了下看起来是可以的
> 
> ```
> tidevice kill com.facebook.WebDriverAgentRunner.xctrunner
> ```

C:\Users\Administrator>tidevice kill com.xxxxxx.sz.xctrunner
Kill pid: 292

C:\Users\Administrator>tidevice kill com.xxxxxx.sz.xctrunner
Kill pid: 292

C:\Users\Administrator>tidevice kill com.xxxxxx.sz.xctrunner
Kill pid: 292

C:\Users\Administrator>tidevice kill com.xxxxxx.sz.xctrunner
Kill pid: 292

C:\Users\Administrator>tidevice kill com.xxxxxx.sz.xctrunner
Kill pid: 292

C:\Users\Administrator>tidevice ps
PID NAME                        BUNDLE_ID                   DISPLAY_NAME
179 GeneralMapsWidget
292 WebDriverAgentRunner-Runner com.screenbeam.sz.xctrunner WebDriverAgentRunner-Runner

包名是自定义的，在使用wda.USBClient()启动服务后，运行tidevice kill com.xxxxxx.sz.xctrunner后返回kill pid:xxx，但是使用tidevice ps查看依旧在运行

NoneTypeCoder, 2023-04-28:
同问 在windows上使用tidevice kill命令好像没有作用。 tidevice是最新的 手机是iPhone12 promax OS 16.1.2

NoneTypeCoder, 2023-04-28:
https://github.com/alibaba/taobao-iphone-device/issues/267