alibaba/tidevice #257: tidevice syslog打印日志时中文是乱码的？

tang-wangyisuifeng, 2022-09-07:
日志打印出来的中文\M-e\M-=\M^S\M-e\M^I\M^M\M-f\M^\\M-

tang-wangyisuifeng, 2022-09-07:
是不是因为系统的原因

Vove7, 2022-11-09:
+1, 在xcode 里显示正常

lycfr, 2022-11-11:
这个问题也发现了，今天才出来

lycfr, 2022-11-22:
```
import sys, io

def decodeSyslog(raw):
    i = 0
    bio = io.BytesIO()
    while i < len(raw):
        c = raw[i:i+1]
        #print(repr(c))
        if c == b'\\':
            if raw[i:].startswith(br'\M^'):
                i += 3
                esc = raw[i]
                i += 1
                bio.write(int.to_bytes((esc & 0x7f) + 0x40, 1, 'little'))
            elif raw[i:].startswith(br'\M-'):
                i += 3
                esc = raw[i]
                i += 1
                bio.write(int.to_bytes((esc & 0x7f) | 0x80, 1, 'little'))
            elif raw[i:].startswith(br'\134'):
                i += 4
                bio.write(b'\\')
            elif raw[i:].startswith(br'\240'):
                i += 4
                bio.write(b'\xa0')
            else:
                #print(raw[i:i+10])
                raise Exception("Fuck1")
        elif c[0] >= 0x20 and c[0] <= 0x7f:
            bio.write(c)
            i += 1
        else:
            raise Exception("Fuck2")
        
    return bio.getvalue()
```
用这个可以解决中文乱码的问题，但是要自己在兼容一下。我本地试了可以

Vove7, 2022-11-22:
@lycfr 👍🏻

codeskyblue, 2022-11-29:
@lycfr 我去，这是什么神奇的转化规则。有没有什么资料参考之类的

lycfr, 2022-12-02:
@lycfr https://gist.github.com/cbracken/d88a84370fdde9cbcfd810d944c8f540

wenboLee, 2023-12-20:
这个怎么用，可以加进来吗