alibaba/tidevice #258: ios15.7 启动wda失败提示tidevice.exceptions.MuxServiceError: InvalidService

wzyzyw, 2022-09-15:
执行以下命令
tidevice wdaproxy -B com.facebook.WebDriverAgentRunner.xctrunner
出现以下报错（看到过类似的问题，说是重启手机，但是我重启后还是报错）
[I 220915 14:19:06 _wdaproxy:125] [00008110-000E0C442182801E] WDA check every 30.0 seconds
[D 220915 14:19:06 _wdaproxy:131] [00008110-000E0C442182801E] launch WDA
[I 220915 14:19:07 _device:969] BundleID: com.facebook.WebDriverAgentRunner.xctrunner
[I 220915 14:19:07 _device:985] ProductVersion: 15.7
[I 220915 14:19:07 _device:986] DeviceIdentifier: 00008110-000E0C442182801E
[I 220915 14:19:07 _device:607] ProductVersion: 15.7
[I 220915 14:19:07 _imagemounter:157] Pushing DeveloperDiskImage.dmg
[I 220915 14:19:07 _imagemounter:170] Push complete
Traceback (most recent call last):
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_device.py", line 492, in start_service
    return self._unsafe_start_service(name)
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_device.py", line 508, in _unsafe_start_service
    raise MuxServiceError(error)
tidevice.exceptions.MuxServiceError: InvalidService

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "d:\software\anaconda\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "d:\software\anaconda\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\__main__.py", line 871, in <module>
    main()
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\__main__.py", line 866, in main
    actions[args.subparser](args)
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\__main__.py", line 259, in cmd_xctest
    d.xctest(args.bundle_id,
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_device.py", line 970, in xctest
    return self.xcuitest(bundle_id, target_bundle_id=target_bundle_id, logger=logger, env=env)
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_device.py", line 996, in xcuitest
    x1 = self._connect_testmanagerd_lockdown()
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_device.py", line 791, in _connect_testmanagerd_lockdown
    conn = self.start_service(
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_device.py", line 494, in start_service
    self.mount_developer_image()
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_device.py", line 668, in mount_developer_image
    self.imagemounter.mount(image_path, signature_path)
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_imagemounter.py", line 139, in mount
    return self.mount_fileobj(image_reader, image_size, signature_content)
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_imagemounter.py", line 182, in mount_fileobj
    self._check_error(ret)
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_imagemounter.py", line 123, in _check_error
    raise MuxError(ret['Error'])
tidevice.exceptions.MuxError: ImageMountFailed
[E 220915 14:19:08 _wdaproxy:150] [00008110-000E0C442182801E] wda started failed
[D 220915 14:19:08 _wdaproxy:131] [00008110-000E0C442182801E] launch WDA
[I 220915 14:19:09 _device:969] BundleID: com.facebook.WebDriverAgentRunner.xctrunner
[I 220915 14:19:09 _device:985] ProductVersion: 15.7
[I 220915 14:19:09 _device:986] DeviceIdentifier: 00008110-000E0C442182801E
[I 220915 14:19:09 _device:607] ProductVersion: 15.7
[I 220915 14:19:09 _imagemounter:157] Pushing DeveloperDiskImage.dmg
[I 220915 14:19:09 _imagemounter:170] Push complete
Traceback (most recent call last):
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_device.py", line 492, in start_service
    return self._unsafe_start_service(name)
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_device.py", line 508, in _unsafe_start_service
    raise MuxServiceError(error)
tidevice.exceptions.MuxServiceError: InvalidService

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "d:\software\anaconda\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "d:\software\anaconda\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
[... 134 lines cut ...]
    exec(code, run_globals)
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\__main__.py", line 871, in <module>
    main()
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\__main__.py", line 866, in main
    actions[args.subparser](args)
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\__main__.py", line 259, in cmd_xctest
    d.xctest(args.bundle_id,
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_device.py", line 970, in xctest
    return self.xcuitest(bundle_id, target_bundle_id=target_bundle_id, logger=logger, env=env)
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_device.py", line 996, in xcuitest
    x1 = self._connect_testmanagerd_lockdown()
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_device.py", line 791, in _connect_testmanagerd_lockdown
    conn = self.start_service(
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_device.py", line 494, in start_service
    self.mount_developer_image()
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_device.py", line 668, in mount_developer_image
    self.imagemounter.mount(image_path, signature_path)
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_imagemounter.py", line 139, in mount
    return self.mount_fileobj(image_reader, image_size, signature_content)
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_imagemounter.py", line 182, in mount_fileobj
    self._check_error(ret)
  File "F:\TET\smart-tools\setup\python_venv\venv\lib\site-packages\tidevice\_imagemounter.py", line 123, in _check_error
    raise MuxError(ret['Error'])
tidevice.exceptions.MuxError: ImageMountFailed
[E 220915 14:19:13 _wdaproxy:150] [00008110-000E0C442182801E] wda started failed

li8911255, 2022-09-15:
同报错，iOS 16 也有同样问题，重启手机、电脑都没用

sunshine2176, 2022-09-19:
iOS16 +1

duanhonggo, 2022-11-28:
ios15.7 +1

duanhonggo, 2022-11-28:
我尝试解决了这个问题，将C:\Users\lenovo\.tidevice\device-support 中 15.7.zip更换就好了，记得要对文件重命名  https://github.com/iGhibli/iOS-DeviceSupport/blob/master/DeviceSupport/15.7(FromXcode_14.1_Release_Candidate_xip).zip

Vantermonde, 2023-03-08:
把手机版本对应的device support添加进去就可以了

ghb0224, 2024-04-24:
> 把手机版本对应的device support添加进去就可以了

ios17.4怎么办，也是同样错