alibaba/tidevice #260: 使用tidevice+wda进行测试一段时间后，部分命令报错ConnectionError: connection closed；

bingosam, 2022-09-27:
# 版本
0.9.11

# 报错ConnectionError的命令（有时报ConnectionResetError: [WinError 10054] 远程主机强迫关闭了一个现有的连接）
- launch
- ps
- wdaproxy
- sysinfo
# 正常的命令
- syslog
- battery
- info
- date
- appinfo
- applist
- screenshot
- uninstall
- install
- reboot
- crashreport
# 错误日志（均类似）
```
C:\Users\Administrator>tidevice wdaproxy -B *.xctrunner
[E 220927 11:22:59 _instruments:594] drain error
    Traceback (most recent call last):
      File "f:\python\anaconda3\lib\site-packages\tidevice\_instruments.py", line 580, in _drain
        self._drain_single_message()
      File "f:\python\anaconda3\lib\site-packages\tidevice\_instruments.py", line 607, in _drain_single_message
        mheader, payload = self.recv_dtx_message()
      File "f:\python\anaconda3\lib\site-packages\tidevice\_instruments.py", line 513, in recv_dtx_message
        message_id = self.recv_part_dtx_message()
      File "f:\python\anaconda3\lib\site-packages\tidevice\_instruments.py", line 409, in recv_part_dtx_message
        data = self.psock.recvall(0x20)
      File "f:\python\anaconda3\lib\site-packages\tidevice\_safe_socket.py", line 113, in recvall
        raise ConnectionError("socket connection broken")
    ConnectionError: socket connection broken
Traceback (most recent call last):
  File "f:\python\anaconda3\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "f:\python\anaconda3\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "f:\python\anaconda3\lib\site-packages\tidevice\__main__.py", line 956, in <module>
    main()
  File "f:\python\anaconda3\lib\site-packages\tidevice\__main__.py", line 951, in main
    actions[args.subparser](args)
  File "f:\python\anaconda3\lib\site-packages\tidevice\__main__.py", line 268, in cmd_xctest
    d.xctest(args.bundle_id,
  File "f:\python\anaconda3\lib\site-packages\tidevice\_device.py", line 967, in xctest
    return self.xcuitest(bundle_id, target_bundle_id=target_bundle_id, logger=logger, env=env)
  File "f:\python\anaconda3\lib\site-packages\tidevice\_device.py", line 1054, in xcuitest
    _, pid = self._launch_app_runner(bundle_id, session_identifier,
  File "f:\python\anaconda3\lib\site-packages\tidevice\_device.py", line 845, in _launch_app_runner
    channel = conn.make_channel(InstrumentsService.ProcessControl)
  File "f:\python\anaconda3\lib\site-packages\tidevice\_instruments.py", line 318, in make_channel
    result = self.call_message(0, '_requestChannelWithCode:identifier:', aux)
  File "f:\python\anaconda3\lib\site-packages\tidevice\_instruments.py", line 363, in call_message
    return self.wait_reply(_id).result
  File "f:\python\anaconda3\lib\site-packages\tidevice\_instruments.py", line 568, in wait_reply
    raise ConnectionError("connection closed")
ConnectionError: connection closed
```

# 初步结论

重启手机后会恢复。但是测试一段时间后又如此。

SuperChaoM, 2022-12-21:
之前是可以的，换了一台电脑,每3个小时左右就掉了，崩溃

SuperChaoM, 2022-12-22:
是不是hub不稳定，我这边2台机器，一台机器正常，一台机器3小时tidevice就找不到设备了,需要重新重启USB或者重新拔插才正常;

SuperChaoM, 2022-12-22:
@bingosam 大佬有解决吗?

xuanhui1991, 2022-12-22:
同样遇到了这个问题，不知道怎么解决

king1782055, 2023-02-10:
切换到~目录下，再执行tidevice命令可以解决该问题。 cd ~.前面在appium-webdriveragent目录

king1782055, 2023-02-10:
我在appium-webdriveragent目录执行tidevice命令会报该错误。切换到~目录下，再执行tidevice命令可以解决该错误。步骤如下：
1.cd ~
2.执行tidevice+wda命令

codeskyblue, 2023-02-23:
没头绪

FatMonkeyy, 2024-11-15:
请问解决了吗，求解决办法

vividius, 2025-03-15:
Please advise. This thing makes me crazy. Happens on 2 different ocmputers and 5 different iphones. Happens randomly.