alibaba/tidevice #269: tidevice 启动 wda 失败

bqwwdsh, 2022-11-08:
报错日志如下：

[I 221107 19:40:17 __init__:245] WDA is not running, exec: /Library/Frameworks/Python.framework/Versions/3.10/bin/tidevice xctest
[I 221107 19:40:18 _device:966] BundleID: com.facebook.WebDriverAgentRunner.xctrunner.xctrunner
[I 221107 19:40:18 _device:982] ProductVersion: 15.7.1
[I 221107 19:40:18 _device:983] DeviceIdentifier: 00008110-001645A63698401E
[I 221107 19:40:18 _device:819] SignIdentity: 'Apple Development: Chenxi Li (V6N3HHBYZS)'
[I 221107 19:40:18 _device:825] CFBundleExecutable: WebDriverAgentRunner-Runner
[I 221107 19:40:18 _device:856] AppContainer: /private/var/mobile/Containers/Data/Application/A9314134-F89A-4AEE-AC60-4E8E4F8C7508
[I 221107 19:40:18 _device:896] Launch 'com.facebook.WebDriverAgentRunner.xctrunner.xctrunner' pid: 53700
[I 221107 19:40:18 _device:1032] Test runner ready detected
[I 221107 19:40:18 _device:1024] Start execute test plan with IDE version: 29
[E 221107 19:40:18 _instruments:594] drain error
    Traceback (most recent call last):
      File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/tidevice/_instruments.py", line 580, in _drain
        self._drain_single_message()
      File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/tidevice/_instruments.py", line 607, in _drain_single_message
        mheader, payload = self.recv_dtx_message()
      File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/tidevice/_instruments.py", line 513, in recv_dtx_message
        message_id = self.recv_part_dtx_message()
      File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/tidevice/_instruments.py", line 409, in recv_part_dtx_message
        data = self.psock.recvall(0x20)
      File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/tidevice/_safe_socket.py", line 113, in recvall
        raise ConnectionError("socket connection broken")
    ConnectionError: socket connection broken
[I 221107 19:40:18 _device:1104] xctrunner quited
[W 221107 19:40:18 _device:923] WebDriverAgentRunner quitted
[W 221107 19:40:20 __init__:255] xctest launch failed

想问一下编译 wda 对 MacOS 和 XCode 版本有要求吗？
手机ios 版本：15.7
MacOS：12.6
XCode：13.2

p.s. 顺便问一下，之前git工程里那个打包好的ipa文件删掉了吗？

myungkyojung, 2022-12-30:
same error!
device is connected with wifi?

codeskyblue, 2023-01-13:
tidevice --trace xctest 试试这个命令呢？

jeniva, 2023-10-19:
> tidevice --trace xctest 试试这个命令呢？

不行，请问还有解决办法吗
[image]

zer0e, 2023-11-13:
请问查到问题所在了吗