alibaba/tidevice #270: 实现类似adbkit的能力

zhangpeibisha, 2022-11-13:
目前iPhone设备比较昂贵，如果公司出现多个办公点，以目前的情况来说，则需要购买多个设备来进行开发和测试。是否可以在iPhone和Pc间加上一个代理服务，使得iPhone设备可以远程被用于xcode调试。

目前我了解到可以模拟lockdown来进行模拟iPhone设备链接usbmuxd，如果我在这个模拟的lockdown提供服务，允许客户端链接这个代理服务，是否就可以完成代理能力实现跨物理设备连接和使用iPhone设备

ZhouYixun, 2022-12-16:
> 目前iPhone设备比较昂贵，如果公司出现多个办公点，以目前的情况来说，则需要购买多个设备来进行开发和测试。是否可以在iPhone和Pc间加上一个代理服务，使得iPhone设备可以远程被用于xcode调试。
> 
> 目前我了解到可以模拟lockdown来进行模拟iPhone设备链接usbmuxd，如果我在这个模拟的lockdown提供服务，允许客户端链接这个代理服务，是否就可以完成代理能力实现跨物理设备连接和使用iPhone设备

sib刚好实现了这个功能 https://sonic-cloud.gitee.io/#/SIB?tag=sib-remote

codeskyblue, 2023-02-23:
利用socat也可以简单的实现一下。

在链接电脑的主机上执行命令

```bash
socat TCP-LISTEN:27015,fork,reuseaddr UNIX:/var/run/usbmuxd
```

在另外一台机器上执行
```bash
tidevice --socket 机器的IP:27015 list
```

ZhouYixun, 2023-02-23:
> 利用socat也可以简单的实现一下。
> 
> 在链接电脑的主机上执行命令
> 
> ```shell
> socat TCP-LISTEN:27015,fork,reuseaddr UNIX:/var/run/usbmuxd
> ```
> 
> 在另外一台机器上执行
> 
> ```shell
> tidevice --socket 机器的IP:27015 list
> ```

学到了

ivoidcat, 2023-03-13:
> 

请问这种方式可以在广域网中使用吗，或者使用vpn