alibaba/tidevice #275: iOS 15.7、iOS 16 及以上版本无法使用部分命令

Vantermonde, 2022-12-27:
caiyun@caiyundeMac-mini ~ % tidevice screenshot screenshot.jpg
Screenshot saved to screenshot.jpg
[I 221227 17:01:28 _device:602] ProductVersion: 16.0.3
[I 221227 17:01:28 _imagemounter:157] Pushing DeveloperDiskImage.dmg
[I 221227 17:01:28 _imagemounter:170] Push complete
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/tidevice/_device.py", line 484, in start_service
    return self._unsafe_start_service(name)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/tidevice/_device.py", line 503, in _unsafe_start_service
    raise MuxServiceError(error)
tidevice.exceptions.MuxServiceError: InvalidService

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/bin/tidevice", line 8, in <module>
    sys.exit(main())
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/tidevice/__main__.py", line 965, in main
    actions[args.subparser](args)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/tidevice/__main__.py", line 274, in cmd_screenshot
    d.screenshot().convert("RGB").save(filename)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/tidevice/_device.py", line 528, in screenshot
    return next(self.iter_screenshot())
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/tidevice/_device.py", line 533, in iter_screenshot
    conn = self.start_service(LockdownService.MobileScreenshotr)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/tidevice/_device.py", line 486, in start_service
    self.mount_developer_image()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/tidevice/_device.py", line 663, in mount_developer_image
    self.imagemounter.mount(image_path, signature_path)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/tidevice/_imagemounter.py", line 139, in mount
    return self.mount_fileobj(image_reader, image_size, signature_content)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/tidevice/_imagemounter.py", line 182, in mount_fileobj
    self._check_error(ret)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/tidevice/_imagemounter.py", line 123, in _check_error
    raise MuxError(ret['Error'])
tidevice.exceptions.MuxError: ImageMountFailed

codeskyblue, 2022-12-30:
[image]

Vantermonde, 2023-01-04:
谢谢你的解决方案，问题搞定了，开发者模式是OK的

closed 2023-01-04 by Vantermonde