alibaba/tidevice #277: tidevice xctest failed!

myungkyojung, 2023-01-17:
I did code signing webdriveragent-runner by **enterprise** **distribution** provision with no errors.
but, when i run xctest using tidevice, it failed.
 ex) tidevice xctest -B my_webdriveragent-runner.xctrunner
i don't know why. let me know any solutions. plz.

I found same issues: https://github.com/alibaba/taobao-iphone-device/issues/14.

InferiorCoder, 2023-03-08:
any bug report can you provide？

myungkyojung, 2023-03-30:
@NoneTypeCoder
This is logogs of the same issue from other people.

[I 210201 21:10:10 _device:912] BundleID: com.xxxxxxxxxx.WebDriverAgentRunner-Runner
[I 210201 21:10:10 _device:914] DeviceIdentifier: 00008030-xxxxxxxxxxxxE
[I 210201 21:10:10 _device:775] SignIdentity: 'iPhone Distribution: xxxxxxxxx Co., Ltd.'
Traceback (most recent call last):
  File "/Users/xx/.pyenv/versions/uiauto8/bin/tidevice", line 8, in <module>
    sys.exit(main())
  File "/Users/xx/.pyenv/versions/3.8.6/envs/uiauto8/lib/python3.8/site-packages/tidevice/__main__.py", line 429, in main
    actions[args.subparser](args)
  File "/Users/xx/.pyenv/versions/3.8.6/envs/uiauto8/lib/python3.8/site-packages/tidevice/__main__.py", line 209, in cmd_xctest
    d.xctest(args.bundle_id, logger=setup_logger(level=logging.INFO), env=env)
  File "/Users/xx/.pyenv/versions/3.8.6/envs/uiauto8/lib/python3.8/site-packages/tidevice/_device.py", line 985, in xctest
    pid = self._launch_wda(bundle_id, session_identifier, env=env, logger=logger)
  File "/Users/xx/.pyenv/versions/3.8.6/envs/uiauto8/lib/python3.8/site-packages/tidevice/_device.py", line 786, in _launch_wda
    for fname in fsync.listdir("/tmp"):
  File "/Users/xx/.pyenv/versions/3.8.6/envs/uiauto8/lib/python3.8/site-packages/tidevice/_sync.py", line 106, in listdir
    pkg = self._recv()
  File "/Users/xx/.pyenv/versions/3.8.6/envs/uiauto8/lib/python3.8/site-packages/tidevice/_sync.py", line 77, in _recv
    buf = buf + self.recvall(FHeader.size - len(buf))
  File "/Users/xx/.pyenv/versions/3.8.6/envs/uiauto8/lib/python3.8/site-packages/tidevice/_safe_socket.py", line 55, in recvall
    raise MuxError("socket connection broken")
tidevice.exceptions.MuxError: socket connection broken

NoneTypeCoder, 2023-03-30:
I maybe know why this issue happened. Can you provide me the version of Xcode, iOS, and WDA which you used?

myungkyojung, 2023-03-30:
@NoneTypeCoder 
Yes, that's right. I used abc.xctrunner as a bundle, is there a solution?

NoneTypeCoder, 2023-03-31:
[image]
I have a guess about this problem， in this step which showed in screenshot，We need to access the /tmp folder，when I added the print statement the printed path is /tmp/WebDriverAgentRunner-D927F243-1159-478F-A429-0F850623BBE9.xctestconfiguration). So we can get access to this file which make this problem happened.
Another related issue https://github.com/libimobiledevice/ifuse/issues/24#issuecomment-93119672.
[image]
[image]
As said above，only work with apps that are deployed via XCode and Ad-Hoc distribution. So thats maybe why we cant get the access from the app that code signing by enterprise distribution.

myungkyojung, 2023-03-31:
@NoneTypeCoder 
I'm trying to deploy an enterprise distribution to install webdriveragent to any devices.
From your answer, it looks like there's no way to do it.

NoneTypeCoder, 2023-03-31:
@myungkyojung Im trying to do this too. In order to solve this problem,  I have tried many ways like modify the Info.plist file, modify the .mobileprovision file or check all the optional options when signing the enterprise. But no one works. Maybe this problem is unsolvable.

NoneTypeCoder, 2023-03-31:
When I modify the Info.plist file and then code signing webdriveragent-runner by enterprise distribution provision. And I modify fsync = self.app_sync(bundle_id, command="VendDocuments") then I get the error below:
AFCPacket(status=<AFCStatus.INVALID_ARG: 7>, data=bytearray(b'\x07\x00\x00\x00\x00\x00\x00\x00'), payload=bytearray(b''))
Traceback (most recent call last):
  File "D:\ENV\lib\site-packages\tidevice\_sync.py", line 328, in _context_open
    yield h
  File "D:\ENV\lib\site-packages\tidevice\_sync.py", line 413, in push_content
    raise MuxError("write error: {!s}".format(pkg.status))
tidevice.exceptions.MuxError: write error: AFCStatus.INVALID_ARG
It seems like I have no access to wirite, compared with before I made these changes, it can be sent and read now. I don't know what I can do anymore.

shensimiao, 2023-08-31:
@NoneTypeCoder Is there a solution now?

hx1997, 2023-10-11:
I came across this problem too, and it turns out if your device is jailbroken and has AFC2 installed, the solution is quite simple.

As NoneTypeCoder noted above, `tidevice` complains about this because it has no access to the `<app container>/tmp` directory. If we're jailbroken, we can easily access it by invoking the AFC2 service. To do this, you simply make a few modifications to the code as detailed below.

1. in `tidevice/_device.py`, look for the `_launch_app_runner` method, and replace the following code:
```
fsync = self.app_sync(bundle_id, command="VendDocuments")

for fname in fsync.listdir("/tmp"):
    if fname.endswith(".xctestconfiguration"):
       logger.debug("remove /tmp/%s", fname)
       fsync.remove("/tmp/" + fname)
fsync.push_content(xctest_path, xctest_content)
```
with
```
fsync = self.sync

for fname in fsync.listdir(app_container + "/tmp"):
    if fname.endswith(".xctestconfiguration"):
       logger.debug("remove /tmp/%s", fname)
       fsync.remove(app_container + "/tmp/" + fname)
fsync.push_content(app_container + xctest_path, xctest_content)
```
2. in `tidevice/_proto.py`, look for the `LockdownService` class, and replace `AFC = "com.apple.afc"` with `AFC = "com.apple.afc2"`

And you're good to go!

shensimiao, 2023-10-16:
but my ios devices is not jailbroken,how can i do?

Guard you（神思者）
***@***.***

Cc: "Guard ***@***.******@***.***&gt;;

 

 
As NoneTypeCoder noted above, tidevice complains about this because it has no access to the <app container&gt;/tmp directory. If we're jailbroken, we can easily access it by invoking the AFC2 service. To do this, you simply make a few modifications to the code as detailed below.