alibaba/tidevice #279: Any plan for springboardservices?

andy380743909, 2023-02-03:
The springboardservices provides many functions to interact with the device, like get/set app icon, get device interface orientation, get wallpaper and so on. https://iphonedev.wiki/index.php/SpringBoard.app/MIG_subsystem

codeskyblue, 2023-02-23:
I searched DevelopImage(16.1), not found service of springboard

```
(base) tidevice ➤ hdiutil mount /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/DeviceSupport/16.
1/DeveloperDiskImage.dmg
Checksumming Whole_Disk…
.......................................................................................................................
                          Whole_Disk: verified   CRC32 $9D79625C
verified   CRC32 $0058A140
/dev/disk2          	                               	/Volumes/DeveloperDiskImage
(base) tidevice ➤ tree /Volumes/DeveloperDiskImage/Library/Lockdown                                           git:main*
/Volumes/DeveloperDiskImage/Library/Lockdown
└── ServiceAgents
    ├── com.apple.GPUTools.MobileService.plist
    ├── com.apple.accessibility.axAuditDaemon.remoteserver.plist
    ├── com.apple.debugserver.DVTSecureSocketProxy.applist.plist
    ├── com.apple.debugserver.DVTSecureSocketProxy.plist
    ├── com.apple.dt.devicearbitration.service.plist
    ├── com.apple.dt.fetchsymbols.plist
    ├── com.apple.dt.simulatelocation.plist
    ├── com.apple.instruments.remoteserver.DVTSecureSocketProxy.plist
    ├── com.apple.mobile.screenshotr.plist
    ├── com.apple.testmanagerd.remote-automation.secure.plist
    └── com.apple.testmanagerd.secure.plist
```

ZhouYixun, 2023-02-23:
https://github.com/SonicCloudOrg/sonic-gidevice/blob/main/pkg/libimobiledevice/springboard.go

Did you mean for `com.apple.springboardservices`  ?

ZhouYixun, 2023-02-23:
In fact, I did use com.apple.springboardservices in my project to obtain app icon, screen rotation and other methods, but my python language level is poor, I don't know if @codeskyblue  has any further plans

andy380743909, 2023-02-25:
> https://github.com/SonicCloudOrg/sonic-gidevice/blob/main/pkg/libimobiledevice/springboard.go
> 
> 
> 
> Did you mean for `com.apple.springboardservices`  ?

Yes! I knew it from the famous imobiledevice project. I am most interested in how do they know the commands that the `com.apple.springboardservices` support, sniffing the usb bus traffic or disassemble the private system frameworks or any other approach?

ZhouYixun, 2023-02-26:
When you use relevant services, you can capture relevant information. You can refer to https://github.com/danielpaulus/go-ios