alibaba/tidevice #280: 求解request error: ('Connection aborted.', MuxReplyError(<UsbmuxReplyCode.ConnectionRefused: 3>))

Sun-GoKu, 2023-02-07:
windows环境
已安装iTunes
wda：https://github.com/appium/WebDriverAgent，已编译安装
tidevice：tidevice version 0.10.1
使用 tidevice -u udid wdaproxy -B Bundle Id --port 8100启动wda时，会有以下报错：
[D 230207 20:41:23 _wdaproxy:58] [00008101-000565382288001E] request error: ('Connection aborted.', MuxReplyError(<UsbmuxReplyCode.ConnectionRefused: 3>))

http://localhost:8100/status有信息正常返回
但http://localhost:8100/inspector 不能查看

brady221018, 2023-02-28:
遇到同样的问题。目前无解。

brady221018, 2023-02-28:
我的问题已解决。TARGET选择不对。在Xcode中，有运行图标的三角符号的那一行，依次应该是WebDriverAgent  WebDriverAgentRunner iPhonexxxxxx。我的问题是第二个选的不对，出问题时我的第二个是IntegrationApp。

LBHHZ, 2023-03-14:
我也出现了这样的问题，重装了一下wda就能用了

codeskyblue, 2023-03-16:
我目前是重启手机

aquasam, 2023-10-20:
相同问题，不过在执行多用例超过1小时后容易出现以上情况，需要等待tidevice重新链接，但执行多用里会上报该错误后停止执行

Zhangronger, 2024-01-15:
也遇到了这个问题，重启重装都不行，求救怎么解决啊
[image]

Zhangronger, 2024-01-16:
iPhone11 iOS14.3遇到同样的报错，求解