alibaba/tidevice #293: OSError: [WinError 10048] 通常每个套接字地址(协议/网络地址/端口)只允许使用一次。

luoxiao1008, 2023-03-10:
D:\download>tidevice list
Traceback (most recent call last):
  File "d:\python36\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "d:\python36\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "D:\Python36\Scripts\tidevice.exe\__main__.py", line 7, in <module>
  File "d:\python36\lib\site-packages\tidevice\__main__.py", line 972, in main
    actions[args.subparser](args)
  File "d:\python36\lib\site-packages\tidevice\__main__.py", line 86, in cmd_list
    ds = um.device_list()
  File "d:\python36\lib\site-packages\tidevice\_usbmux.py", line 80, in device_list
    data = self.send_recv(payload, timeout=10)
  File "d:\python36\lib\site-packages\tidevice\_usbmux.py", line 52, in send_recv
    s = self.create_connection()
  File "d:\python36\lib\site-packages\tidevice\_usbmux.py", line 48, in create_connection
    psock = PlistSocket(self.__address, self._next_tag())
  File "d:\python36\lib\site-packages\tidevice\_safe_socket.py", line 156, in __init__
    super().__init__(addr)
  File "d:\python36\lib\site-packages\tidevice\_safe_socket.py", line 75, in __init__
    self._sock.connect(addr)
OSError: [WinError 10048] 通常每个套接字地址(协议/网络地址/端口)只允许使用一次。

luoxiao1008, 2023-03-10:
经常在执行过程中就报错
************connection error happen****************
('Connection aborted.', OSError(10048, '通常每个套接字地址(协议/网络地址/端口)只允许使用一次。', None, 10048, None))
然后自己敲tidevice命令也会报错

codeskyblue, 2023-03-16:
换linux或mac吧

effort-top, 2025-05-30:
大佬解决了吗