alibaba/tidevice #294: 为什么 ios 版本大于 15.7 时就要调用 enable_ios16_developer_mode 方法

bqwwdsh, 2023-03-11:
[image]

为什么 ios 版本大于 15.7 时就要调用 enable_ios16_developer_mode 方法 
这会导致 ios 版本为 15.7 的 iphone 13 在执行 launch app 命令时报错

Traceback (most recent call last):
  File "C:\Python310\lib\site-packages\tidevice\_device.py", line 501, in start_service
    return self._unsafe_start_service(name)
  File "C:\Python310\lib\site-packages\tidevice\_device.py", line 520, in _unsafe_start_service
    raise MuxServiceError(error)
tidevice.exceptions.MuxServiceError: InvalidService

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Python310\Scripts\tidevice.exe\__main__.py", line 7, in <module>
  File "C:\Python310\lib\site-packages\tidevice\__main__.py", line 972, in main
    actions[args.subparser](args)
  File "C:\Python310\lib\site-packages\tidevice\__main__.py", line 342, in cmd_launch
    with d.connect_instruments() as ts:
  File "C:\Python310\lib\site-packages\decorator.py", line 232, in fun
    return caller(func, *(extras + args), **kw)
  File "C:\Python310\lib\site-packages\retry\api.py", line 73, in retry_decorator
    return __retry_internal(partial(f, *args, **kwargs), exceptions, tries, delay, max_delay, backoff, jitter,
  File "C:\Python310\lib\site-packages\retry\api.py", line 33, in __retry_internal
    return f()
  File "C:\Python310\lib\site-packages\tidevice\_device.py", line 817, in connect_instruments
    conn = self.start_service(
  File "C:\Python310\lib\site-packages\tidevice\_device.py", line 503, in start_service
    self.mount_developer_image()
  File "C:\Python310\lib\site-packages\tidevice\_device.py", line 667, in mount_developer_image
    self.enable_ios16_developer_mode()
  File "C:\Python310\lib\site-packages\tidevice\_device.py", line 471, in enable_ios16_developer_mode
    is_developer = self.get_value("DeveloperModeStatus", domain="com.apple.security.mac.amfi")
  File "C:\Python310\lib\site-packages\tidevice\_device.py", line 372, in get_value
    return ret['Value']
KeyError: 'Value'

Vantermonde, 2023-03-16:
[image]
改成15.7.2就可以了

codeskyblue, 2023-03-16:
哈，这样吗？我改改

closed 2023-03-16 by codeskyblue

NoneTypeCoder, 2023-03-22:
是的 目前os 15.x 最高版本是15.7  ios16才需要启动开发者模式 不改为15.7.2（改为16也可以）的话 就会让15.7也去启动开发者模式就有问题。 请问啥时候会将这个改动release到pypi上？

codeskyblue, 2023-03-22:
0.10.3忘记发布了

Liuluu, 2023-04-17:
建议直接改成16
15.7.2的设备又因为这个跑不起来了

NoneTypeCoder, 2023-04-17:
@Liuluu 我也遇到15.7.2因为这个报错跑不起来 然后我将源码改成16成功跑起来后 再将源码改为15.7.2貌似又能直接跑了 有点奇怪。 你可以试试。同建议改为16

starry-ww, 2023-04-17:
> @Liuluu 我也遇到15.7.2因为这个报错跑不起来 然后我将源码改成16成功跑起来后 再将源码改为15.7.2貌似又能直接跑了 有点奇怪。 你可以试试。同建议改为16

我直接把源码改成16了。确实有点奇怪，因为同样的手机（15.7.2）在另外的电脑上是正常的，没深究过原因。