alibaba/tidevice #299: Drain error while launching your WebDriverAgent.ipa

EmpireST1, 2023-03-17:
Hello. I trying to run Appium IOS + Windows debugging.

But now i feel issue when i trying to run com.facebook.WebDriverAgentRunner

`[E 230317 20:55:55 _instruments:621] drain error
    Traceback (most recent call last):
      File "c:\users\empire\appdata\local\programs\python\python38-32\lib\site-packages\tidevice\_instruments.py", line 607, in _drain
        self._drain_single_message()
      File "c:\users\empire\appdata\local\programs\python\python38-32\lib\site-packages\tidevice\_instruments.py", line 634, in _drain_single_message
        mheader, payload = self.recv_dtx_message()
      File "c:\users\empire\appdata\local\programs\python\python38-32\lib\site-packages\tidevice\_instruments.py", line 540, in recv_dtx_message
        message_id = self.recv_part_dtx_message()
      File "c:\users\empire\appdata\local\programs\python\python38-32\lib\site-packages\tidevice\_instruments.py", line 436, in recv_part_dtx_message
        data = self.psock.recvall(0x20)
      File "c:\users\empire\appdata\local\programs\python\python38-32\lib\site-packages\tidevice\_safe_socket.py", line 115, in recvall
        raise ConnectionError("socket connection broken")
    ConnectionError: socket connection broken`

Can u share me new WebDriverAgent.ipa?

closed 2023-03-19 by EmpireST1

AliceHome, 2023-03-21:
这个问题是解决了吗？我今天也遇到了

EmpireST1, 2023-03-21:
> 到

Partially, issue on webdriveragent.ipa
You should to try different webdrivers. On Mac OS same issue with this webdriveragent.ipa

jeniva, 2023-10-19:
请问解决了吗，我也遇到这个错误了

jeniva, 2023-10-19:
[image]