alibaba/tidevice #301: start wda on iphone5s with ios 12.5.7 failed (ConnectionError: socket connection broken)[在12.5.7的iphone5s上启动wda失败]

gawinwong, 2023-03-27:
#### Environment
- ios: 12.5.7
- device: 2 iphone 5s (Jailbroken and Unjailbroken)
- macos: 12.6.3 Monterey
- xcode: 14.3
- python: 3.7 && 3.9
- node: 12.22.12
- tidevice: 0.10.3
- profile: person development

It is usable with basic commands such as `list`, `applist`, `launch` and so on, but WDA cannot be started through USB with the command `tidevice xctest -B com.xxx.xxx.xxxx`, the  error `ConnectionError: socket connection broken` will be reported. The devices have been bound to the profile. I have changed the version of node and python, but it still cannot be started. The wda.ipa with the same profile works fine on iphone6s 15.7.1.

`list`, `applist`, `launch` 等基本命令可以使用，但是WDA不能通过USB使用`tidevice xctest -B com.xxx.xxx.xxxx`启动，会报错`ConnectionError: socket connection broken` 。 设备已绑定配置文件。 更改了node和python的版本，还是无法启动。同样profile的`wda.ipa`在iphone6s 15.7.1上倒是没问题。

gawinwong, 2023-03-27:
[image]
 
update to 0.10.4, it changes to this error `tidevice.exceptions.SocketError: recvall: socket connection broken`

AINYRolls, 2024-09-10:
How did you solve this problem?