alibaba/tidevice #312: How do I convert PIL python image to NSImage?

dhiravidamani238, 2023-04-25:
Hi

I'm getting value like "<PIL.PngImagePlugin.PngImageFile image mode=RGB size=349x500 at 0x103BB1390>". How do i convert this value to NSImage using swift language in Xcode tool. 

If there is any python plugin available?

Please reply..

codeskyblue, 2023-05-04:
If you want to convert a PIL image to an NSImage in Python, you can use the Cocoa bindings library provided by PyObjC.

Here's an example code that shows how to convert a PIL image to an NSImage:

python
from PIL import Image
import io
from AppKit import NSImage, NSData, NSBitmapImageRep, NSPasteboard, NSTIFFPboardType

# Load the image
pil_image = Image.open("my_image.png")

# Convert the PIL image to a data buffer
with io.BytesIO() as buffer:
    pil_image.save(buffer, format="PNG")
    data = buffer.getvalue()

# Create an NSImage from the data buffer
image_data = NSData.dataWithBytes_length_(data, len(data))
ns_image = NSImage.alloc().initWithData_(image_data)

In the above code, we first load the PIL image and save it to a data buffer in PNG format. We then create an NSData object from the data buffer and use it to initialize an NSImage object. The resulting NSImage object can be used to display the image in a macOS application.

Note that before you can use the PyObjC library, you need to install it using the following command:

bash
pip install pyobjc-framework-Cocoa

I hope this helps!
END

From chatgpt  https://chat.devsleep.com

dhiravidamani238, 2023-05-04:
> If you want to convert a PIL image to an NSImage in Python, you can use the Cocoa bindings library provided by PyObjC.
>
> Here's an example code that shows how to convert a PIL image to an NSImage: [...]

@codeskyblue Thank you