alibaba/tidevice #323: windows上wdaproxy报未知异常，超时

tester-chang, 2023-06-12:
windows
version：0.9.12（最新也试过）

日志如下：`C:\Users\xxx>tidevice -u 00008030-0005650E2E7A802E wdaproxy -B com.facebook.WebDriverAgentRunner.xctrunner --port 8100
[I 230612 19:14:48 _wdaproxy:128] [00008030-0005650E2E7A802E] WDA check every 30.0 seconds
[D 230612 19:14:48 _wdaproxy:134] [00008030-0005650E2E7A802E] launch WDA
[W 230612 19:14:49 _wdaproxy:65] [00008030-0005650E2E7A802E] Unknown exception: Timeout value connect was Timeout(connect=90, read=90, total=None), but it must be an int, float or None.
[I 230612 19:14:49 _device:968] BundleID: com.facebook.WebDriverAgentRunner.xctrunner
[I 230612 19:14:49 _device:984] ProductVersion: 15.6.1
[I 230612 19:14:49 _device:985] DeviceIdentifier: 00008030-0005650E2E7A802E
[I 230612 19:14:49 _device:821] SignIdentity: 'Apple Development: XXXXXXXX'
[I 230612 19:14:49 _device:827] CFBundleExecutable: WebDriverAgentRunner-Runner
[I 230612 19:14:50 _device:858] AppContainer: /private/var/mobile/Containers/Data/Application/6DCF4890-3C34-4CF9-82FA-5C18AF5183E0
[I 230612 19:14:50 _device:898] Launch 'com.facebook.WebDriverAgentRunner.xctrunner' pid: 870
[I 230612 19:14:50 _device:1034] Test runner ready detected
[I 230612 19:14:50 _device:1026] Start execute test plan with IDE version: 29
[W 230612 19:14:50 _wdaproxy:65] [00008030-0005650E2E7A802E] Unknown exception: Timeout value connect was Timeout(connect=90, read=90, total=None), but it must be an int, float or None.
[I 230612 19:14:50 _device:934] WebDriverAgent start successfully
[W 230612 19:14:51 _wdaproxy:65] [00008030-0005650E2E7A802E] Unknown exception: Timeout value connect was Timeout(connect=90, read=90, total=None), but it must be an int, float or None.
[W 230612 19:14:52 _wdaproxy:65] [00008030-0005650E2E7A802E] Unknown exception: Timeout value connect was Timeout(connect=90, read=90, total=None), but it must be an int, float or None.`

tester-chang, 2023-06-12:
[W 230612 19:14:52 _wdaproxy:65] [00008030-0005650E2E7A802E] Unknown exception: Timeout value connect was Timeout(connect=90, read=90, total=None), but it must be an int, float or None.

tester-chang, 2023-06-12:
已解决，是requests的版本不对，升级即可，升级后版本为2.31.0
pip install --upgrade requests

huajianlin20230911, 2023-10-30:
大佬牛逼，已经解决