alibaba/tidevice #332: 多次launch会导致timeout，The handshake operation timed out

wjwABCDEFG, 2023-07-13:
我尝试使用com.apple.springboard亮屏，自动化测试在大约经过400条的测试用例后，会出现如下错误，需要重启手机
`C:\Users\wjw>tidevice launch com.apple.springboard
Traceback (most recent call last):
  File "C:\Users\wjw\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\wjw\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\wjw\AppData\Local\Programs\Python\Python310\Scripts\tidevice.exe\__main__.py", line 7, in <module>
  File "C:\Users\wjw\AppData\Local\Programs\Python\Python310\lib\site-packages\tidevice\__main__.py", line 978, in main
    actions[args.subparser](args)
  File "C:\Users\wjw\AppData\Local\Programs\Python\Python310\lib\site-packages\tidevice\__main__.py", line 341, in cmd_launch
    with d.connect_instruments() as ts:
  File "C:\Users\wjw\AppData\Local\Programs\Python\Python310\lib\site-packages\decorator.py", line 232, in fun
    return caller(func, *(extras + args), **kw)
  File "C:\Users\wjw\AppData\Local\Programs\Python\Python310\lib\site-packages\retry\api.py", line 73, in retry_decorator
    return __retry_internal(partial(f, *args, **kwargs), exceptions, tries, delay, max_delay, backoff, jitter,
  File "C:\Users\wjw\AppData\Local\Programs\Python\Python310\lib\site-packages\retry\api.py", line 33, in __retry_internal
    return f()
  File "C:\Users\wjw\AppData\Local\Programs\Python\Python310\lib\site-packages\tidevice\_device.py", line 818, in connect_instruments
    conn = self.start_service(
  File "C:\Users\wjw\AppData\Local\Programs\Python\Python310\lib\site-packages\tidevice\_device.py", line 502, in start_service
    return self._unsafe_start_service(name)
  File "C:\Users\wjw\AppData\Local\Programs\Python\Python310\lib\site-packages\tidevice\_device.py", line 541, in _unsafe_start_service
    conn = self.create_inner_connection(data['Port'], _ssl=_ssl, ssl_dial_only=ssl_dial_only)
  File "C:\Users\wjw\AppData\Local\Programs\Python\Python310\lib\site-packages\tidevice\_device.py", line 292, in create_inner_connection
    psock.switch_to_ssl(self.ssl_pemfile_path)
  File "C:\Users\wjw\AppData\Local\Programs\Python\Python310\lib\site-packages\tidevice\_safe_socket.py", line 162, in switch_to_ssl
    ssock = context.wrap_socket(self._sock, server_hostname="iphone.localhost")
  File "C:\Users\wjw\AppData\Local\Programs\Python\Python310\lib\ssl.py", line 513, in wrap_socket
    return self.sslsocket_class._create(
  File "C:\Users\wjw\AppData\Local\Programs\Python\Python310\lib\ssl.py", line 1071, in _create
    self.do_handshake()
  File "C:\Users\wjw\AppData\Local\Programs\Python\Python310\lib\ssl.py", line 1342, in do_handshake
    self._sslobj.do_handshake()
TimeoutError: _ssl.c:980: The handshake operation timed out`

wjwABCDEFG, 2023-07-14:
更新到0.10.11后，报错改为connection closed，应该和https://github.com/alibaba/taobao-iphone-device/issues/320是一样的

jackhmy, 2023-08-14:
遇到相同的问题；
`while true; do tidevice --udid <#udid#> launch <#bundleID#>; tidevice --udid <#udid#> kill <#bundleID#>; done;!`

在终端执行上述命令，循环启停，大概300-400次后会出现：
> Traceback (most recent call last):
  File "/opt/homebrew/bin/tidevice", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/tidevice/__main__.py", line 978, in main
    actions[args.subparser](args)
  File "/opt/homebrew/lib/python3.11/site-packages/tidevice/__main__.py", line 342, in cmd_launch
    pid = ts.app_launch(args.bundle_id,
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/tidevice/_instruments.py", line 703, in app_launch
    code = self.make_channel(self._SERVICE_PROCESS_CONTROL)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/tidevice/_instruments.py", line 344, in make_channel
    result = self.call_message(0, '_requestChannelWithCode:identifier:', aux)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/tidevice/_instruments.py", line 389, in call_message
    return self.wait_reply(_id).result
           ^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/tidevice/_instruments.py", line 594, in wait_reply
    raise ConnectionError("connection closed")
ConnectionError: connection closed