alibaba/tidevice #337: iOS 17 无法运行

li8911255, 2023-07-27:
目前 iOS 17 无法正常运行，提示缺少  device support 文件
[image]

按照苹果社区官方人员的说法（https://developer.apple.com/forums/thread/730947） iOS 17 之后不再提供 device support 文件
请问这块有考虑做相关的兼容吗

sunshine2176, 2023-08-08:
同问，iOS17怎么处理

yanglikai0806, 2023-08-23:
同问，如何解决

songjt1993, 2023-08-30:
同问

lycfr, 2023-08-30:
https://github.com/doronz88/pymobiledevice3  这个支持iOS17

yiqianghe, 2023-09-05:
同问

li8911255, 2023-09-12:
@codeskyblue 正式版马上要发了，大佬这块有进展吗

idanhauser, 2023-09-19:
iOS17 will be supported?

marrash, 2023-12-13:
會針對iOS17進行更新嗎

codeskyblue, 2023-12-19:
正在看，目前发现有一个项目 pymobiledevice3已经支持在mac上运行了，可以参考一下。我目前正在看他的代码

gjs1990717, 2023-12-26:
> 正在看，目前发现有一个项目 pymobiledevice3已经支持在mac上运行了，可以参考一下。我目前正在看他的代码

咋样了？

gjs1990717, 2023-12-27:
咋样了？？

gjs1990717, 2023-12-28:
？

gjs1990717, 2024-01-09:
这个算是凉凉？

codeskyblue, 2024-01-11:
本来想自己实现iOS17的支持，但是发现自己没这个实力，不过社区还是很强大的，目前发现了一个好项目能支持iOS17： pymobiledevice3，这个项目的代码写的非常好，因为太过优秀，我就不继续在tidevice里面去实现了。不过这个项目的命令行好像不太友好，所以我又搞了一个新项目

https://github.com/codeskyblue/tidevice3

基于pymobiledevice3做封装，封装了一些常用的功能，如app install uninstall screenshot。欢迎体验

至于XCUITest的支持，还要在等等。pr还没写完

saildf, 2024-08-16:
> 本来想自己实现iOS17的支持，但是发现自己没这个实力，不过社区还是很强大的，目前发现了一个好项目能支持iOS17： pymobiledevice3，这个项目的代码写的非常好，因为太过优秀，我就不继续在tidevice里面去实现了。不过这个项目的命令行好像不太友好，所以我又搞了一个新项目
> 
> https://github.com/codeskyblue/tidevice3
> 
> 基于pymobiledevice3做封装，封装了一些常用的功能，如app install uninstall screenshot。欢迎体验
> 
> 至于XCUITest的支持，还要在等等。pr还没写完

咋样了大佬，等支持XCTEST\用xcodebuild启动wda太慢了