alibaba/tidevice #343: Windows系统上，tidevice启动wda后直接quitted

freefunfee7, 2023-08-15:
Mac系统版本：Ventura 13.4
Xcode版本：15.0 beta 6
iPhone系统版本：16.0.2
WebDriverAgent的代码库，git pull到了最新内容（2023.8.15）

（1）首先使用Mac的Xcode安装了Webdriveragent-runner 到iPhone手机上，可以正常test成功
（2）接下来换到Windows电脑上，Windows电脑上的tidevice版本：0.11.0 Python版本：3.9.10
（3）执行命令：tidevice xctest -B com.facebook.WebDriverAgentRunner.dream.xctrunner
提示WebDriverAgentRunner quitted
[image]

（4）执行命令：tidevice xctest -B com.facebook.WebDriverAgentRunner.dream.xctrunner --debug
[image]

以上，在Windows系统上，tidevice启动wda后直接quitted，无法成功。
期待您的回复，谢谢

freefunfee7, 2023-08-17:
我又安装了Xcode 14.3.1版本，其他的环境版本没有调整。
然后基于Xcode 14.3.1将Webdriveragent-runner 到iPhone手机上，可以正常test成功。
并且之后执行命令tidevice xctest -B com.facebook.WebDriverAgentRunner.dream.xctrunner
可以正常运行。

基于以上的操作，不同点只在于Xcode 的版本
Xcode15 beta 6

SurelySomeday, 2023-08-20:
老哥，可以提供你编译好的ipa文件吗？我没有MAC环境编译

tewbie, 2023-09-26:
xcode15 正式版。和楼主一样的问题。启动直接WebDriverAgentRunner quitted

SimonChou-17LIVE, 2023-10-12:
同样问题, xcode 15 无法正常启动 WDA, xcode 14.x 版本可以

SmallMousePP, 2023-10-12:
同样问题, xcode 15 无法正常启动 WDA, xcode 14.x 版本可以

codeskyblue, 2023-10-19:
0.12.0已修复，试试看

SimonChou-17LIVE, 2023-11-10:
更新v0.12.0, 可以正常启动 WDA