alibaba/tidevice #345: tidevice.exceptions.ServiceError: ApplicationVerificationFailed

pppppino, 2023-08-28:
我按照 https://github.com/alibaba/taobao-iphone-device/blob/main/DEVELOP.md#how-to-package-wdaipa 生成了 WDA.ipa, 并且可以在一台 iPhone （iphone8. 15.4.1）上安装， 但是在另外一台 iPhone （iPhone X 15.6.1）上安装就不行, 会报错：
```
[E 230828 17:12:38 _installation:56] ApplicationVerificationFailed
[E 230828 17:12:38 _installation:57] Failed to verify code signature of /var/installd/Library/Caches/com.apple.mobile.installd.staging/temp.i6TFll/extracted/Payload/WebDriverAgentRunner-Runner.app : 0xe8008015 (A valid provisioning profile for this executable was not found.)
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.10/bin/tidevice", line 8, in <module>
    sys.exit(main())
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/tidevice/__main__.py", line 972, in main
    actions[args.subparser](args)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/tidevice/__main__.py", line 164, in cmd_install
    bundle_id = d.app_install(args.filepath_or_url)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/tidevice/_device.py", line 794, in app_install
    return self.installation.install(bundle_id, target_path)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/tidevice/_installation.py", line 58, in install
    raise ServiceError(progress['Error'])
tidevice.exceptions.ServiceError: ApplicationVerificationFailed
```

Quietly-20201113, 2023-09-28:
我的打包成功 安装失败 
```
Install: VerifyingApplication (40%)ERROR: Install failed. Got error "ApplicationVerificationFailed" with code 0xe8008015: Failed to verify code signature of /var/installd/Library/Caches/com.apple.mobile.installd.staging/temp.hfLlfA/extracted/Payload/HBuilder.app : 0xe8008015 (A valid provisioning profile for this executable was not found.)
```

codeskyblue, 2023-10-19:
应该是证书签名问题，没有将另外一台设备增加到 mobileprovision里面去

closed 2023-10-19 by codeskyblue