alibaba/tidevice #349: Multiple functionalities are broken on iOS 17

mhaoli, 2023-09-25:
Apple has developed a new device connectivity stack to target devices running iOS 17 and later, which has affected many third-party iOS device controlling tools (e.g. libimobiledevice).

I have tried following Tidevice functionalities and found some of them are broken:

<b style="font-weight:normal;" id="docs-internal-guid-40c1129d-7fff-b9af-05f6-b720942c949f"><div dir="ltr" style="margin-left:0pt;" align="left">

tool | Does it support iOS 17
-- | --
screenshot | ❌
launch | ❌
kill | ❌
ps | ❌
relay | ? (no error raised)
fsync(ls, rm, push, pull, stat) | ✅
applist | ✅
install | ✅
uninstall | ✅
syslog | ✅
</div></b>

mhaoli, 2023-09-25:
Recording some information that might be helpful:

* Looks like iOS 17 is using a new device connectivity stack (named CoreDevice), changes involves:
  * [communication protocol](https://github.com/doronz88/pymobiledevice3/blob/master/misc/RemoteXPC.md)
  * A github [PR](https://github.com/doronz88/pymobiledevice3/pull/463/files) in pymobiledevice3 to support mounting "new" developer images on iOS 17. 
* A libimobiledevice [issue](https://github.com/libimobiledevice/libimobiledevice/issues/1452) that records some investigation. Looks like this [comment](https://github.com/libimobiledevice/libimobiledevice/issues/1452#issuecomment-1638859453) has fixed a part of controlling features in pymobiledevice3
* go-ios [issue](https://github.com/danielpaulus/go-ios/issues/265)
* appium [issues](https://github.com/appium/appium/issues/18749)

kinget007, 2023-11-22:
lanuch: xcrun devicectl device process launch

mhaoli, 2023-12-22:
Yes, Apple released the new tool `devicectl` to support **some** of the functionalities. But we still expect that tidevice can support them because:

1. We need to use them on Linux.
2. We are using `tidevice` now and it would be good if we could use one tool for all functionalities instead of depending on multiple tools...

lezhi12, 2024-01-10:
ios 17没有device support这个问题，perfdog貌似已经解决了，现在用perfdog可以采集ios17得性能，哪位老师/同学，知道perfdog大概是通过什么思路解决得这个问题么？

idonen, 2024-01-10:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

codeskyblue, 2024-01-12:
iOS17 support will be continue in repo https://github.com/codeskyblue/tidevice3