alibaba/tidevice #366: 获取手机列表报错，20多台手机的时候会出现这个错误4294967295 is not a valid UsbmuxReplyCode

418040413, 2024-01-13:
Traceback (most recent call last):
  File "d:\py\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "d:\py\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "D:\py\Scripts\tidevice.exe\__main__.py", line 7, in <module>
  File "d:\py\lib\site-packages\tidevice\__main__.py", line 1027, in main
    actions[args.subparser](args)
  File "d:\py\lib\site-packages\tidevice\__main__.py", line 84, in cmd_list
    ds = um.device_list()
  File "d:\py\lib\site-packages\tidevice\_usbmux.py", line 80, in device_list
    data = self.send_recv(payload, timeout=10)
  File "d:\py\lib\site-packages\tidevice\_usbmux.py", line 54, in send_recv
    self._check(data)
  File "d:\py\lib\site-packages\tidevice\_usbmux.py", line 98, in _check
    raise MuxReplyError(data['Number'])
  File "d:\py\lib\site-packages\tidevice\exceptions.py", line 25, in __init__
    self.reply_code = UsbmuxReplyCode(number)
  File "d:\py\lib\enum.py", line 315, in __call__
    return cls.__new__(cls, value)
  File "d:\py\lib\enum.py", line 569, in __new__
    raise exc
  File "d:\py\lib\enum.py", line 553, in __new__
    result = cls._missing_(value)
  File "d:\py\lib\enum.py", line 582, in _missing_
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: 4294967295 is not a valid UsbmuxReplyCode

418040413, 2024-01-13:
是windows系统

83715405, 2024-03-01:
大佬，解决了么？我也遇到这个问题