alibaba/tidevice #368: [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] request error: ('Connection aborted.', MuxReplyError(<UsbmuxReplyCode.ConnectionRefused: 3>))

eechenyi, 2024-01-26:
tidevice版本  tidevice version 0.12.7
iPhone 8  Ios版本16.7
在linux的ubuntu20.0的服务器上，运行tidevice -u *** wdaproxy  -p 8100，
运行一段时间后出现错误，且重启服务器，重启手机，重装wda都不行，但换个电脑是可以的。报错信息如下：
 tidevice -u af999665d5f7e8e75f09ab4d80fe5719d1859eb9 wdaproxy -p 23431
[I 240126 10:03:02 _wdaproxy:128] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] WDA check every 30.0 seconds
[D 240126 10:03:02 _wdaproxy:134] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] launch WDA
[D 240126 10:03:02 _wdaproxy:58] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] request error: ('Connection aborted.', MuxReplyError(<UsbmuxReplyCode.ConnectionRefused: 3>))
[I 240126 10:03:02 _device:996] BundleID: com.facebook.WebDriverAgentRunner.xctrunner
[I 240126 10:03:02 _device:1023] ProductVersion: 15.7
[I 240126 10:03:02 _device:1024] UDID: af999665d5f7e8e75f09ab4d80fe5719d1859eb9
[I 240126 10:03:03 _device:857] SignIdentity: 'Apple Development: zhiju ren (4R8AL7A5H4)'
[I 240126 10:03:03 _device:863] CFBundleExecutable: WebDriverAgentRunner-Runner
[D 240126 10:03:03 _wdaproxy:58] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] request error: ('Connection aborted.', MuxReplyError(<UsbmuxReplyCode.ConnectionRefused: 3>))
[I 240126 10:03:03 _device:929] Launch 'com.facebook.WebDriverAgentRunner.xctrunner' pid: 29806
[I 240126 10:03:03 _device:1072] Test runner ready detected
[I 240126 10:03:03 _device:1064] Start execute test plan with IDE version: 29
[I 240126 10:03:03 _device:1190] xctrunner quited with result:
    
[W 240126 10:03:03 _device:937] WebDriverAgentRunner quitted
[D 240126 10:03:04 _wdaproxy:58] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] request error: ('Connection aborted.', MuxReplyError(<UsbmuxReplyCode.ConnectionRefused: 3>))
[E 240126 10:03:04 _wdaproxy:153] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] wda started failed
[D 240126 10:03:04 _wdaproxy:134] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] launch WDA
[D 240126 10:03:04 _wdaproxy:58] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] request error: ('Connection aborted.', MuxReplyError(<UsbmuxReplyCode.ConnectionRefused: 3>))
[I 240126 10:03:04 _device:996] BundleID: com.facebook.WebDriverAgentRunner.xctrunner
[I 240126 10:03:04 _device:1023] ProductVersion: 15.7
[I 240126 10:03:04 _device:1024] UDID: af999665d5f7e8e75f09ab4d80fe5719d1859eb9
[I 240126 10:03:05 _device:857] SignIdentity: 'Apple Development: zhiju ren (4R8AL7A5H4)'
[I 240126 10:03:05 _device:863] CFBundleExecutable: WebDriverAgentRunner-Runner
[D 240126 10:03:05 _wdaproxy:58] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] request error: ('Connection aborted.', MuxReplyError(<UsbmuxReplyCode.ConnectionRefused: 3>))
[I 240126 10:03:05 _device:929] Launch 'com.facebook.WebDriverAgentRunner.xctrunner' pid: 29808
[I 240126 10:03:05 _device:1072] Test runner ready detected
[I 240126 10:03:05 _device:1064] Start execute test plan with IDE version: 29
[I 240126 10:03:05 _device:1190] xctrunner quited with result:
    
[W 240126 10:03:05 _device:937] WebDriverAgentRunner quitted
[D 240126 10:03:06 _wdaproxy:58] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] request error: ('Connection aborted.', MuxReplyError(<UsbmuxReplyCode.ConnectionRefused: 3>))
[E 240126 10:03:06 _wdaproxy:153] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] wda started failed
[D 240126 10:03:06 _wdaproxy:134] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] launch WDA
[D 240126 10:03:06 _wdaproxy:58] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] request error: ('Connection aborted.', MuxReplyError(<UsbmuxReplyCode.ConnectionRefused: 3>))
[I 240126 10:03:06 _device:996] BundleID: com.facebook.WebDriverAgentRunner.xctrunner
[I 240126 10:03:06 _device:1023] ProductVersion: 15.7
[I 240126 10:03:06 _device:1024] UDID: af999665d5f7e8e75f09ab4d80fe5719d1859eb9
[I 240126 10:03:07 _device:857] SignIdentity: 'Apple Development: zhiju ren (4R8AL7A5H4)'
[I 240126 10:03:07 _device:863] CFBundleExecutable: WebDriverAgentRunner-Runner
[D 240126 10:03:07 _wdaproxy:58] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] request error: ('Connection aborted.', MuxReplyError(<UsbmuxReplyCode.ConnectionRefused: 3>))
[I 240126 10:03:07 _device:929] Launch 'com.facebook.WebDriverAgentRunner.xctrunner' pid: 29810
[I 240126 10:03:07 _device:1072] Test runner ready detected
[I 240126 10:03:07 _device:1064] Start execute test plan with IDE version: 29
[I 240126 10:03:07 _device:1190] xctrunner quited with result:
    
[D 240126 10:03:08 _wdaproxy:58] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] request error: ('Connection aborted.', MuxReplyError(<UsbmuxReplyCode.ConnectionRefused: 3>))
[E 240126 10:03:08 _wdaproxy:153] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] wda started failed
[D 240126 10:03:08 _wdaproxy:134] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] launch WDA
[D 240126 10:03:08 _wdaproxy:58] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] request error: ('Connection aborted.', MuxReplyError(<UsbmuxReplyCode.ConnectionRefused: 3>))
[I 240126 10:03:08 _device:996] BundleID: com.facebook.WebDriverAgentRunner.xctrunner
[I 240126 10:03:08 _device:1023] ProductVersion: 15.7
[I 240126 10:03:08 _device:1024] UDID: af999665d5f7e8e75f09ab4d80fe5719d1859eb9
[I 240126 10:03:09 _device:857] SignIdentity: 'Apple Development: zhiju ren (4R8AL7A5H4)'
[I 240126 10:03:09 _device:863] CFBundleExecutable: WebDriverAgentRunner-Runner
[D 240126 10:03:09 _wdaproxy:58] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] request error: ('Connection aborted.', MuxReplyError(<UsbmuxReplyCode.ConnectionRefused: 3>))
[I 240126 10:03:09 _device:929] Launch 'com.facebook.WebDriverAgentRunner.xctrunner' pid: 29812
[I 240126 10:03:09 _device:1072] Test runner ready detected
[I 240126 10:03:09 _device:1064] Start execute test plan with IDE version: 29
[I 240126 10:03:09 _device:1190] xctrunner quited with result:
    
[W 240126 10:03:09 _device:937] WebDriverAgentRunner quitted
[D 240126 10:03:10 _wdaproxy:58] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] request error: ('Connection aborted.', MuxReplyError(<UsbmuxReplyCode.ConnectionRefused: 3>))
[E 240126 10:03:10 _wdaproxy:153] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] wda started failed
[D 240126 10:03:10 _wdaproxy:134] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] launch WDA
[D 240126 10:03:10 _wdaproxy:58] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] request error: ('Connection aborted.', MuxReplyError(<UsbmuxReplyCode.ConnectionRefused: 3>))
[I 240126 10:03:10 _device:996] BundleID: com.facebook.WebDriverAgentRunner.xctrunner
[I 240126 10:03:10 _device:1023] ProductVersion: 15.7
[I 240126 10:03:10 _device:1024] UDID: af999665d5f7e8e75f09ab4d80fe5719d1859eb9
[I 240126 10:03:11 _device:857] SignIdentity: 'Apple Development: zhiju ren (4R8AL7A5H4)'
[I 240126 10:03:11 _device:863] CFBundleExecutable: WebDriverAgentRunner-Runner
[D 240126 10:03:11 _wdaproxy:58] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] request error: ('Connection aborted.', MuxReplyError(<UsbmuxReplyCode.ConnectionRefused: 3>))
[I 240126 10:03:11 _device:929] Launch 'com.facebook.WebDriverAgentRunner.xctrunner' pid: 29815
[I 240126 10:03:11 _device:1072] Test runner ready detected
[I 240126 10:03:11 _device:1064] Start execute test plan with IDE version: 29
[I 240126 10:03:11 _device:1190] xctrunner quited with result:
    
[D 240126 10:03:12 _wdaproxy:58] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] request error: ('Connection aborted.', MuxReplyError(<UsbmuxReplyCode.ConnectionRefused: 3>))
[E 240126 10:03:12 _wdaproxy:153] [af999665d5f7e8e75f09ab4d80fe5719d1859eb9] wda started failed

iricox, 2024-03-08:
请问这个错误，你解决了吗，我也遇到了这个问题

wangyuqi2017, 2024-04-12:
没有解决，重启手机电脑都不行

wangyuqi2017, 2024-04-12:
> 请问这个错误，你解决了吗，我也遇到了这个问题

没有解决，重启手机、电脑都不行

jianghuihon, 2024-05-20:
我也遇到这个问题，是tidevice的问题吗
[image]